liquidctl/liquidctl #750: Corsair Commander Core XT Fans Stuck at 100% After Updates

chrispouliot, 2024-12-13:
### Describe the bug

I updated my system today (Silverblue Fedora 41) and I have been having errors from CoolerControl originating from LiquidCTL and my fans connected to my Commander Core XT have been stuck at 100%.
Before today all fans worked properly with fan curves set to the GPU temp.

### Commands executed

```sudo liquidctl set fan1 speed 10 --debug```

### Output of all relevant commands with `--debug` flag

```/usr/lib/python3.13/site-packages/docopt.py:165: SyntaxWarning: invalid escape sequence '\S'
  name = re.findall('(<\S*?>)', source)[0]
/usr/lib/python3.13/site-packages/docopt.py:166: SyntaxWarning: invalid escape sequence '\['
  value = re.findall('\[default: (.*)\]', source, flags=re.I)
/usr/lib/python3.13/site-packages/docopt.py:207: SyntaxWarning: invalid escape sequence '\['
  matched = re.findall('\[default: (.*)\]', description, flags=re.I)
/usr/lib/python3.13/site-packages/docopt.py:456: SyntaxWarning: invalid escape sequence '\S'
  split = re.split('\n *(<\S+?>|-\S+?)', doc)[1:]
[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.11.11-300.fc41.x86_64-x86_64-with-glibc2.40
[DEBUG] (cli) (_log_env_infos): python: 3.13.0 (main, Oct  8 2024, 00:00:00) [GCC 14.2.1 20240912 (Red Hat 14.2.1-3)]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.2
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0.post2
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 11.0.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: 10-0050 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-MSFT8000:00 (skipped, invalid literal for int() with base 10: 'MSFT8000:00')
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core XT (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xffbc usage=0x0088)
[DEBUG] (usb) (find_devices): HID device: 046d:4074 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:4074 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4074 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4074 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4074 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4074 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4074 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c53f (usage_page=0xff00 usage=0x0002)
[... 50 lines cut ...]
  File "/usr/lib/python3.13/site-packages/liquidctl/driver/commander_core.py", line 207, in _read_data
    raise ExpectationNotMet('device returned incorrect data type')
liquidctl.error.ExpectationNotMet: unspecified liquidctl error
[ERROR] (cli) (log): Corsair Commander Core XT (experimental): unspecified liquidctl error
```

### Affected device

Corsair Commander Core XT

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Fedora 41 Kernel 6.11.11.300

### Installation method

dnf install (through rpm-ostree)

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.11.11-300.fc41.x86_64-x86_64-with-glibc2.40)

jonasmalacofilho, 2024-12-13:
This seems to be just another instance of #583.

However, since it apparently only started to happen after an update, can you please check your logs and find out the exact versions of the Fedora liquidctl packages before and after this latest update that may have triggered the issue?

(Still, it could just a coincidence).

chrispouliot, 2024-12-13:
What’s confusing to me is it’s worked perfectly for the last month despite the bug that matches this being quite old. I’ll try and debug tomorrow (I’m out all day today) and see what I can rollback. Thanks!

jonasmalacofilho, 2024-12-16:
Merged with other similar issues into #753.

closed 2024-12-16 by jonasmalacofilho