liquidctl/liquidctl #753: Corsair Commander Core/Core XT/ST firmware 2.0 issues

jonasmalacofilho, 2024-12-16:
### Describe the bug

The liquidctl driver has issues with Corsair Commander Core, Core XT and ST models running firmware version 2.

The most common sign is hitting an `ExpectationNotMet` exception.

```
[INFO] (cli) (log): detailed error: Corsair Commander Core XT (experimental): unspecified liquidctl error: ExpectationNotMet('device returned incorrect data type')
Traceback (most recent call last):
  File "/usr/lib/python3.13/site-packages/liquidctl/cli.py", line 475, in main
    _device_set_speed(dev, args, **opts)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/liquidctl/cli.py", line 294, in _device_set_speed
    dev.set_fixed_speed(args['<channel>'].lower(), int(args['<percentage>'][0]), **opts)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/liquidctl/driver/commander_core.py", line 147, in set_fixed_speed
    res = self._read_data(_MODE_HW_SPEED_MODE, _DATA_TYPE_HW_SPEED_MODE)
  File "/usr/lib/python3.13/site-packages/liquidctl/driver/commander_core.py", line 207, in _read_data
    raise ExpectationNotMet('device returned incorrect data type')
liquidctl.error.ExpectationNotMet: unspecified liquidctl error
[ERROR] (cli) (log): Corsair Commander Core XT (experimental): unspecified liquidctl error
```

---

This issues merges and replaces #520, #583, #598, #623, #729, #750. Check those for more information.

Additionally, the following issues *may* also be related: #705

### Commands executed

(See original issues)

### Output of all relevant commands with `--debug` flag

(See original issues)

### Affected device

Corsair Commander Core, Core XT and ST models

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Unrelated

### Installation method

Unrelated

### Version of liquidctl

liquidctl v1.13.0

CrackerJackMack, 2025-01-06:
I don't know if this will help or is a known "fix".  The errors seemed similar enough. I was going to get device captures by running windows 11 VM, USB made available to VM. iCue sees the device correctly. I had previously been able to set fixed fan speed with liquidctl just fine (don't care about rgb right now). Then I started to get bad data back as `ERROR: Corsair Commander Core XT (experimental): unspecified liquidctl error`. The captures weren't very helpful, tried wireshark in both linux and windows, seemed to provide minimal info. After changing some speeds in iCue and shutting down the VM, I can set all the fans again! I've rebooted multiple times and I forgot to save `--debug` output. But now that I have everything setup, I should be pretty easy to re-get captures. My guess, and maybe already known, is that the state of the device gets in an invalid state. 

Additionally, a thing I also did was update the on-device memory settings via iCue. Maybe that's what fixed it instead of setting the running fan speeds? Let me know if I can help or if this was even helpful.

```
Firmware 0.14.62
Commander CORE XT
```

```
kevin@Kevin-mint:~/liquidctl$ lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 22
Release:	22
Codename:	wilma
```

```
kevin@Kevin-mint:~/liquidctl$ liquidctl list
Device #0: ASUS Aura LED Controller
Device #1: Corsair Commander Core XT (experimental)
kevin@Kevin-mint:~/liquidctl$ lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 22
Release:	22
Codename:	wilma
```
```
kevin@Kevin-mint:~/liquidctl$ liquidctl --version
liquidctl v0.0.0-unknown (Linux-6.8.0-51-generic-x86_64-with-glibc2.39)
```
```
kevin@Kevin-mint:~/liquidctl$ apt show liquidctl
Package: liquidctl
Version: 1.13.0-2
Priority: optional
Section: universe/admin
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Laszlo Boszormenyi (GCS) <gcs@debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 431 kB
Depends: python3-colorlog, python3-crcmod, python3-docopt, python3-hid, python3-pil, python3-smbus, python3-usb, python3:any
Homepage: https://github.com/liquidctl/liquidctl
Download-Size: 83.0 kB
APT-Manual-Installed: yes
APT-Sources: http://babylon.cs.uh.edu/mirror-sites/ubuntu noble/universe amd64 Packages
Description: CLI and Python drivers for AIO liquid coolers and other devices
 liquidctl is a tool for controlling various settings of PC internals, such as:
  [ trimmed ]
```

CrackerJackMack, 2025-01-06:
oh ballsack. In more testing I broke it again. I'll trying fiddling and take better notes this time.

CrackerJackMack, 2025-02-03:
Well, I can confirm that it's related to the stored memory state on the device. Full power-off and on cycle with like 15 minutes in between had no changes. There is some magic combo to get the onboard memory back into alignment. I would guess we need to analyze the state more to see what switches need to be flipped.

ingenlabsltd, 2025-04-13:
Any fix for this problem? Damn my fans is all spinning at around 1000rpm .

TheSHEEEP, 2025-05-02:
> Any fix for this problem? Damn my fans is all spinning at around 1000rpm .

My "solution" was to buy a different fan controller (Aqua Computer Quadro in my case)... probably not what anyone was hoping for, but it seemed the most logical and least time intensive fix. Aqua Computer stuff generally has very good Linux support.

jluizpjr, 2025-07-29:
Another weird situation. Started a VM with USB passthrough and XT worked as expected but, after shutting down the VM and rebooting the host liquidctl is back to normal. Following a couple debugs:

===============[ERROR]=====================
root@pve:~# liquidctl set --match XT fan1 speed 10 --debug
[DEBUG] (cli) (_log_env_infos): script: /usr/local/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.15.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.8.12-12-pve-x86_64-with-glibc2.36
[DEBUG] (cli) (_log_env_infos): python: 3.11.2 (main, Apr 28 2025, 14:11:48) [GCC 12.2.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.9.0.post3
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1.post2
[DEBUG] (cli) (_log_env_infos): with pillow: 9.4.0
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1.post2
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3 (SMBus PIIX4 adapter port 1 at 0b20)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1 (SMBus PIIX4 adapter port 0 at 0b00)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-CYP0002:00 skipped: invalid literal for int() with base 10: 'CYP0002:00'
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2 (SMBus PIIX4 adapter port 2 at 0b00)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0 (Synopsys DesignWare I2C adapter)
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AsusRyujin, AuraLed, CommanderCore, CommanderPro, Coolit, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, MpgCooler, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0xffbc usage=0x0088)
[DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 0b05:19af (usage_page=0xff72 usage=0x00a1)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core XT (broken)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 05e3:0608
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 10c4:ea60
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1b1c:0c2a
[DEBUG] (usb) (find_devices): USB device: 0b05:19af
[DEBUG] (usb) (find_devices): USB device: 046d:c534
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[... 205 lines cut ...]
[DEBUG] (usb) (clear_enqueued_reports): discarded 0 previously enqueued reports
[... cut ...] 0:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] (usb) (read): read 96 bytes: 00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
=========================[END NORMAL RUN]====================================

jluizpjr, 2025-07-29:
It doesn't make any sense to me. I'm trying to replicate the error can everything is just fine, even after a 10 min power off removing power cable.
A clue...  while starting iCUE I saw an "updating" message under commander core but since it didn't update the firmware neither iCUE SW/driver I almost disregarded... until now.

jluizpjr, 2025-07-29:
> It doesn't make any sense to me. I'm trying to replicate the error can everything is just fine, even after a 10 min power off removing power cable. A clue... while starting iCUE I saw an "updating" [...]

Following iCUE log .... you can see even iCUE is having problems... but in the end iCUE is working normally

```
2025-07-29 13:35:23.184 I cue.crashreporthandler.init: begin initialize CrashHandler
2025-07-29 13:35:23.278 I cue.crashreporthandler.init: Started CrashHandler.
2025-07-29 13:35:25.637 I cue.init: iCUE version: 5.31.112
2025-07-29 13:35:25.637 I cue.init: iCUE module version: "5.31.112"
2025-07-29 13:35:25.637 I cue.init: Build type: Public Release
2025-07-29 13:35:25.637 I cue.init: Application ID: 10704
2025-07-29 13:35:25.637 I cue.init: iCUE safe mode disable
2025-07-29 13:35:25.637 I cue.api.websocket.server: Starting...
2025-07-29 13:35:25.653 I cue.api.websocket.server: Started WEB server at port 50123
2025-07-29 13:35:25.653 I cue.init: Previous iCUE version: "5.31.112"
2025-07-29 13:35:25.653 I cue.qml.plugins.loader: Loading qml-plugins from "C:/Program Files/Corsair/Corsair iCUE5 Software/cue_qml_plugins" ; 58 items present.
2025-07-29 13:35:25.653 I cue.qml.plugins.loader: Loading qml-plugin "ActionsCueQmlPlugin.dll"
2025-07-29 13:35:25.669 I cue.qml.plugins.loader: Loading qml-plugin "ActuationCueQmlPlugin.dll"
2025-07-29 13:35:25.684 I cue.qml.plugins.loader: Loading qml-plugin "AlertsCueQmlPlugin.dll"
2025-07-29 13:35:25.700 I cue.qml.plugins.loader: Loading qml-plugin "CalibrationCueQmlPlugin.dll"
2025-07-29 13:35:25.700 I cue.qml.plugins.loader: Loading qml-plugin "CanvasCueQmlPlugin.dll"
2025-07-29 13:35:25.715 I cue.qml.plugins.loader: Loading qml-plugin "CanvasEffectsCueQmlPlugin.dll"
2025-07-29 13:35:25.731 I cue.qml.plugins.loader: Loading qml-plugin "CoolingCueQmlPlugin.dll"
2025-07-29 13:35:25.731 I cue.qml.plugins.loader: Loading qml-plugin "CuestyleCueQmlPlugin.dll"
2025-07-29 13:35:25.793 I cue.qml.plugins.loader: Loading qml-plugin "CurveEditorsCueQmlPlugin.dll"
2025-07-29 13:35:25.809 I cue.qml.plugins.loader: Loading qml-plugin "DashboardCueQmlPlugin.dll"
2025-07-29 13:35:25.825 I cue.qml.plugins.loader: Loading qml-plugin "DashboardScreenCueQmlPlugin.dll"
2025-07-29 13:35:25.825 I cue.qml.plugins.loader: Loading qml-plugin "DeckCueQmlPlugin.dll"
2025-07-29 13:35:25.840 I cue.qml.plugins.loader: Loading qml-plugin "DeclarationsCueQmlPlugin.dll"
2025-07-29 13:35:25.856 I cue.qml.plugins.loader: Loading qml-plugin "DeviceViewCueQmlPlugin.dll"
2025-07-29 13:35:25.872 I cue.qml.plugins.loader: Loading qml-plugin "DialCueQmlPlugin.dll"
2025-07-29 13:35:25.887 I cue.qml.plugins.loader: Loading qml-plugin "DpiCueQmlPlugin.dll"
2025-07-29 13:35:25.887 I cue.qml.plugins.loader: Loading qml-plugin "DramCueQmlPlugin.dll"
2025-07-29 13:35:25.903 I cue.qml.plugins.loader: Loading qml-plugin "EqualizerCueQmlPlugin.dll"
2025-07-29 13:35:25.919 I cue.qml.plugins.loader: Loading qml-plugin "GamepadCueQmlPlugin.dll"
2025-07-29 13:35:25.919 I cue.qml.plugins.loader: Loading qml-plugin "GiphyCueQmlPlugin.dll"
2025-07-29 13:35:25.934 I cue.qml.plugins.loader: Loading qml-plugin "GoveeCueQmlPlugin.dll"
2025-07-29 13:35:25.950 I cue.qml.plugins.loader: Loading qml-plugin "GraphCueQmlPlugin.dll"
2025-07-29 13:35:25.965 I cue.qml.plugins.loader: Loading qml-plugin "HelpCueQmlPlugin.dll"
2025-07-29 13:35:25.965 I cue.qml.plugins.loader: Loading qml-plugin "LaptopCoolingCueQmlPlugin.dll"
2025-07-29 13:35:25.981 I cue.qml.plugins.loader: Loading qml-plugin "LegacySimpleCueQmlPlugin.dll"
2025-07-29 13:35:25.997 I cue.qml.plugins.loader: Loading qml-plugin "LightingsCueQmlPlugin.dll"
2025-07-29 13:35:25.997 I cue.qml.plugins.loader: Loading qml-plugin "LoginCueQmlPlugin.dll"
2025-07-29 13:35:26.012 I cue.qml.plugins.loader: Loading qml-plugin "MainInstallerCueQmlPlugin.dll"
2025-07-29 13:35:26.028 I cue.qml.plugins.loader: Loading qml-plugin "MainWindowPlugin.dll"
2025-07-29 13:35:26.043 I cue.qml.plugins.loader: Loading qml-plugin "MimicCueQmlPlugin.dll"
2025-07-29 13:35:26.043 I cue.qml.plugins.loader: Loading qml-plugin "ModelsCueQmlPlugin.dll"
2025-07-29 13:35:26.059 I cue.qml.plugins.loader: Loading qml-plugin "NanoleafCueQmlPlugin.dll"
2025-07-29 13:35:26.075 I cue.qml.plugins.loader: Loading qml-plugin "NotificationsCueQmlPlugin.dll"
2025-07-29 13:35:26.090 I cue.qml.plugins.loader: Loading qml-plugin "NvMaxineCueQmlPlugin.dll"
2025-07-29 13:35:26.090 I cue.qml.plugins.loader: Loading qml-plugin "PerformanceCueQmlPlugin.dll"
2025-07-29 13:35:26.106 I cue.qml.plugins.loader: Loading qml-plugin "PhilipsHueCueQmlPlugin.dll"
2025-07-29 13:35:26.122 I cue.qml.plugins.loader: Loading qml-plugin "PictureSettingsCueQmlPlugin.dll"
2025-07-29 13:35:26.122 I cue.qml.plugins.loader: Loading qml-plugin "ProfilesCueQmlPlugin.dll"
2025-07-29 13:35:26.137 I cue.qml.plugins.loader: Loading qml-plugin "QuickLightingZonesCueQmlPlugin.dll"
2025-07-29 13:35:26.153 I cue.qml.plugins.loader: Loading qml-plugin "ScreenEffectsCueQmlPlugin.dll"
2025-07-29 13:35:26.153 I cue.qml.plugins.loader: Loading qml-plugin "SettingsCueQmlPlugin.dll"
2025-07-29 13:35:26.168 I cue.qml.plugins.loader: Loading qml-plugin "SimpleCueQmlPlugin.dll"
2025-07-29 13:35:26.184 I cue.qml.plugins.loader: Loading qml-plugin "SlipstreamCueQmlPlugin.dll"
2025-07-29 13:35:26.200 I cue.qml.plugins.loader: Loading qml-plugin "StylePartsCueQmlPlugin.dll"
[... cut ...]
2025-07-29 13:36:32.354 W qt.qml.signalhandler: Property 'onClicked' of object QQuickItemDelegate_QML_585(0x23a8aeab1d0) is a signal handler. You should not call it directly. Make it a proper function and call that or emit the signal.
2025-07-29 13:36:32.601 I cue.actions.voicemod.controller: Connection to server lost.
2025-07-29 13:36:32.601 I cue.actions.voicemod.controller: Connection to server lost.
2025-07-29 13:36:34.932 W qt.qml.signalhandler: Property 'onClicked' of object QQuickItemDelegate_QML_585(0x23a8aeb4070) is a signal handler. You should not call it directly. Make it a proper function and call that or emit the signal.

```

jluizpjr, 2025-07-29:
> Another weird situation. Started a VM with USB passthrough and XT worked as expected in ICUE but, after shutting down the VM and rebooting the host liquidctl is back to normal. Following a couple debugs:
> 
> ```
> ===============[ERROR]=====================
> root@pve:~# liquidctl set --match XT fan1 speed 10 --debug
> [DEBUG] (cli) (_log_env_infos): script: /usr/local/bin/liquidctl
> [DEBUG] (cli) (_log_env_infos): version: 1.15.0
> [DEBUG] (cli) (_log_env_infos): platform: Linux-6.8.12-12-pve-x86_64-with-glibc2.36
> [DEBUG] (cli) (_log_env_infos): python: 3.11.2 (main, Apr 28 2025, 14:11:48) [GCC 12.2.0]
> [DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
> [DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
> [DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
> [DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
> [DEBUG] (cli) (_log_env_infos): with hidapi: 0.9.0.post3
> [DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1.post2
> [DEBUG] (cli) (_log_env_infos): with pillow: 9.4.0
> [DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
> [DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
> [DEBUG] (cli) (_log_env_infos): with smbus: 1.1.post2
> [DEBUG] (smbus) (find_devices): searching LinuxI2c
> [DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
> [DEBUG] (smbus) (find_devices): I²C adapter: i2c-3 (SMBus PIIX4 adapter port 1 at 0b20)
> [DEBUG] (smbus) (find_devices): I²C adapter: i2c-1 (SMBus PIIX4 adapter port 0 at 0b00)
> [DEBUG] (smbus) (find_devices): I²C adapter: i2c-CYP0002:00 skipped: invalid literal for int() with base 10: 'CYP0002:00'
> [DEBUG] (smbus) (find_devices): I²C adapter: i2c-2 (SMBus PIIX4 adapter port 2 at 0b00)
> [DEBUG] (smbus) (find_devices): I²C adapter: i2c-0 (Synopsys DesignWare I2C adapter)
> [DEBUG] (usb) (find_devices): searching HidapiBus
> [DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AsusRyujin, AuraLed, CommanderCore, CommanderPro, Coolit, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, MpgCooler, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
> [DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0x0001 usage=0x0006)
> [DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0x0001 usage=0x0006)
> [DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0x000c usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0x0001 usage=0x0080)
> [DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0xffbc usage=0x0088)
> [DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0xff00 usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0xff00 usage=0x0002)
> [DEBUG] (usb) (find_devices): HID device: 046d:4023 (usage_page=0xff00 usage=0x0004)
> [DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0x0001 usage=0x0002)
> [DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0x0001 usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0x000c usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0xff00 usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0xff00 usage=0x0002)
> [DEBUG] (usb) (find_devices): HID device: 046d:4058 (usage_page=0xff00 usage=0x0004)
> [DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0x0001 usage=0x0002)
> [DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0x0001 usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0x000c usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0x0001 usage=0x0080)
> [DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0xff00 usage=0x0001)
> [DEBUG] (usb) (find_devices): HID device: 046d:c534 (usage_page=0xff00 usage=0x0002)
> [DEBUG] (usb) (find_devices): HID device: 0b05:19af (usage_page=0xff72 usage=0x00a1)
> [DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0001)
> [DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core XT (broken)
> [DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0002)
> [DEBUG] (usb) (find_devices): searching PyUsbBus
> [DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0003
> [DEBUG] (usb) (find_devices): USB device: 05e3:0608
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0002
> [DEBUG] (usb) (find_devices): USB device: 10c4:ea60
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0002
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0003
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0003
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0002
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0003
> [DEBUG] (usb) (find_devices): USB device: 1b1c:0c2a
> [DEBUG] (usb) (find_devices): USB device: 0b05:19af
> [DEBUG] (usb) (find_devices): USB device: 046d:c534
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0002
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0003
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0002
> [DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[... 212 lines cut ...]
[... cut ...] 0:00:00:00:00:00:00:00:00:00
> [DEBUG] (usb) (read): read 96 bytes: 00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> =========================[END NORMAL RUN]====================================
> ```

[image]
 
Left a successful call and Right and error call

randomizedcoder, 2025-12-18:
Just order a "Corsair Commander PRO" and forget about it.

Took a moment to get fan2go reading the temp on the GPU, and then adjusting the Corsair Commander PRO, but now it works great.  Here's the nix for it.
```
#
# fan2go.nix
#
# Not controlling the Radeon Pro W5700, because lact is doing this
# The config controls the corsair fan1 based on the temperature of the Radeon Pro VII/MI50
#
# sudo systemctl status fan2go
# sudo journalctl -u fan2go --follow
# sudo ls /var/lib/fan2go
#
# NATIVE KERNEL DRIVER IMPLEMENTATION:
# 1. setPwm: Directly writes PWM value (0-255) to /sys/class/hwmon/hwmon7/pwm1
# 2. getPwm: Reads current PWM value from /sys/class/hwmon/hwmon7/pwm1
# 3. getRpm: Reads current RPM from /sys/class/hwmon/hwmon7/fan1_input
# 4. Uses native corsair-cpro kernel driver (no liquidctl needed)
# 5. Direct sysfs interface - no external tools required
# 6. More reliable and faster than liquidctl
#
# Nix string literals:
# https://nix.dev/manual/nix/2.28/language/string-literals.html#string-literals
# https://nix.dev/manual/nix/2.28/language/string-interpolation.html
# https://github.com/NixOS/nix/blob/master/doc/manual/source/language/string-interpolation.md
#
# See also: https://github.com/arnarg/config/blob/8de65cf5f1649a4fe6893102120ede4363de9bfa/hosts/terra/fan2go.nix
#
# https://www.kernel.org/doc/html/latest/hwmon/corsair-cpro.html
#
# [  133.368198] corsair-cpro 0003:1B1C:0C10.0006: hidraw5: USB HID v1.11 Device [Corsair Commander PRO] on usb-0000:04:00.3-4/input0
#
# Native kernel driver detected at: /sys/class/hwmon/hwmon7/
# - fan1_input: Current RPM reading
# - pwm1: PWM control (0-255)
# - fan1_label: "fan1 4pin"
#
{
  lib,
  config,
  pkgs,
  ...
}:
let

  cfg = config.hardware.fan2go;

  # Path to the Corsair Commander PRO hwmon device
  # Note: hwmon device numbers can change after kernel updates
  # Check with: ls -la /sys/class/hwmon/ and find the one named "corsaircpro"
  corsairHwmonPath = "/sys/class/hwmon/hwmon5";

  # Debug level for the scripts (0=off, 7=max debug).
  debugLevel = 7;

  # fan2go monitoring script using Prometheus metrics
  fan2goMonitorScript = pkgs.writeShellApplication {
    name = "fan2go-monitor.bash";
    runtimeInputs = [ pkgs.curl pkgs.gawk pkgs.coreutils ];
    text = ''
      #!/bin/bash
      # fan2go monitoring script using Prometheus metrics
      # Provides real-time status of fan control system

      METRICS_URL="http://localhost:9900/metrics"

      # Colors for output
      RED='\033[0;31m'
      GREEN='\033[0;32m'
[... 610 lines cut ...]
        #RestrictAddressFamilies = [ "AF_INET" "AF_INET6" ];
        #Delegate = false;

        LimitNOFILE = 128;
        #LimitNOFILE = 8192;

        # Create and manage the state directory
        StateDirectory = "fan2go";

        # Allow access to hwmon devices for fan control
        # The service needs to read/write to /sys/class/hwmon/hwmon5/pwm1
        # Note: sysfs requires special handling - we need full access to /sys
        PrivateDevices = false;
        ProtectKernelTunables = false;
        # Don't restrict /sys - we need full read/write access for hwmon control
        # ReadWritePaths with /sys/class/hwmon should work, but we may need the parent /sys too
        ReadWritePaths = [
          "/sys"
        ];

      };
    };
  };
}
```

indyfive11, 2026-04-18:
Hi all — I've been running a Commander ST under CoolerControl and hit most of the bugs described in this thread. After USB protocol captures and hardware testing, I submitted #886 which addresses several root causes.

**What's actually happening**

There turn out to be four distinct bugs in the current fw2.x driver code, each manifesting differently:

The most visible one — random `AssertionError('response does not match command')` — is a race condition. fw2.x devices send spontaneous unsolicited HID reports while the driver is waiting for a command response. If one arrives at exactly the right moment, it has the correct report number but is a stale echo, and the hard assert fires. This explains why it's intermittent and gets worse under load (as @CrackerJackMack and @randomizedcoder described).

The `ExpectationNotMet('device returned incorrect data type')` errors have two separate causes. On a factory-fresh device, or after another application (OpenLinkHub, a VM, etc.) has cleared the device state, the speed-mode endpoint returns all zeros. The driver's read-verify guard refuses to write to an uninitialized endpoint, so every fan channel fails on first use — this is the "device gets into an invalid state" scenario that a full power-off or iCUE interaction would sometimes reset. Separately, fw2.x shifted the fan speed endpoint indices on bus `0x6d` relative to fw1.x, so without a firmware branch, fan commands go to the wrong slots and are silently ignored by the device.

The fourth bug is specific to systems running LED animation under CPU load: animation frame writes can exceed 400ms and hold a HID serialization lock, blocking fan control operations past the liqctld request deadline.

**Technical summary**

PR #886 (`commander_core.py`, fw2.x code paths only — fw1.x is untouched):

1. **fw2.x endpoint map** — Added `_is_fw2` branch in `set_fixed_speed` / `set_speed_profile` using the correct endpoint offsets for fw2.x. Verified on Commander ST fw2.0.19: fan1 618 → 2286 RPM after the fix; 2-minute stress test confirmed all channels tracked a temperature profile.

2. **HID desync in `_send_command()`** — Retry up to 8 times, skipping mismatched-echo reports. The original assert is preserved so genuine protocol errors still surface. Eliminates the intermittent crash without masking real failures.

3. **Uninitialized speed-mode endpoint** — At the end of `initialize()` (fw2.x only), if the speed-mode endpoint reads as `(0x00, 0x00)`, seed it with a default payload via `_write_data_bootstrap()`. The device retains the value in NVRAM — bootstrap runs once and is skipped on subsequent starts.

4. **Animation lock timeout** — Replace the blocking `_anim_lock` acquire in `_wake_device_context()` with `acquire(timeout=0.4)`. If the timeout fires, the animation thread is stopped (cosmetic) so the fan operation proceeds (safety-critical). Moved `_save_led_state()` outside the lock — file I/O has no business holding a HID serialization lock.

The PR is tested on Commander ST fw2.0.19. Since this issue tracks all fw2.x devices (Core, Core XT, ST), bugs 2 and 3 in particular may also help Core XT users on fw2.x firmware — though I can't verify that without hardware.

randomizedcoder, 2026-04-18:
@indyfive11 Nice stuff!  Thanks for digging into it!

toledkrw, 2026-04-18:
thanks for the detailed write-up in #886 — Personally, I'd like to refrain that 100% ai-bashing-its-head-into-problem, specially on hardware issues, which are frail. 
As per see, you brought valuable stuff around the driver itself, and branching it for fw1 and fw2,  it aided me in my approach for solving this issue for my Commander Core XT (fw 1.4.62) (I know this gh issues are about fw2, but, this technical issue is also seen on fw1).

Here's what I found after cross-referencing your changes with my own USB captures (**usbmon,
iCUE running in a Windows VM**) and live hardware testing on interface 0:

## 1-  about the Stale Response Bug: confirmed in fw1.x, same fix (retries) applies.
### The race condition exists in fw1.x too, but I believe the trigger is slightly different:
 a kernel HID URB completing *between* `clear_enqueued_reports()` and the subsequent `read()`
 call, placing a stale echo in the buffer before our write response arrives. The symptom is
 identical — `res[1] != command[0]` — just caused by a leftover kernel report rather than a
 spontaneous device report.
 
 The retry-loop fix from your PR applies directly. I applied it with a limit of 10 iterations
 and `raise ExpectationNotMet` instead of `assert`, consistent with the rest of the driver; `initialize()` now passes cleanly every time.

## 2- Channel 0 for reads — confirmed OK on interface 0
### My hypothesis "B" (channel 0 being wrong) does not apply to reads on interface 0. 
Opening endpoint `0x60 0x6d` with `(0x0d, 0x00)` and reading works correctly. The device responds and sends back the expected payload.

However... my captures show iCUE always uses channel 1 (`0x0d 0x01`) for *writes* on interface 1!
Whether interface 0 requires the same (with pagination) is not yet confirmed — the write path crashes before
reaching `_write_data`, so this is still an open question for me.

## 3- New finding: wrong data type constants for fw1.x — this is blocking set_fixed_speed

After fixing the retry loop, `set_fixed_speed` crashes at the *first* `_read_data` call with
`ExpectationNotMet('device returned incorrect data type')`. The debug trace shows:

```
# Open endpoint `0x60 0x6d` (MODE_HW_SPEED_MODE), canal 0
write: 08:0d:00:60:6d:00...
read:  00:0d:00:...         ← ok

# READ_INITIAL
write: 08:08:00:01:00...
read:  00:08:00:7e:20:09:00:00:00:01:ff:00:00:00:01:
----------------^^^^^ raw_data[3:5] = (0x7e, 0x20)
```

The driver checks `raw_data[3:5] != _DATA_TYPE_HW_SPEED_MODE` and throws. The current
constants `(0x03, 0x00)`, `(0x04, 0x00)`, `(0x05, 0x00)` seem to be all wrong for this device.

Cross-referencing with my iCUE captures (case1 scan, fw1.4.62):

| Endpoint   | Driver constant            | Actual dtype  | Source         |
|------------|---------------------------|---------------|----------------|
| `0x60 0x6d` | `_DATA_TYPE_HW_SPEED_MODE = (0x03, 0x00)`  | `(0x7e, 0x20)` | hardware confirmed |
| `0x61 0x6d` | `_DATA_TYPE_HW_FIXED_PERCENT = (0x04, 0x00)` | `(0x7e, 0x20)` | "case1" capture |
| `0x62 0x6d` | `_DATA_TYPE_HW_CURVE_PERCENT = (0x05, 0x00)` | `(0x03, 0x00)` | "case1" capture |

(case 1 stands for "start_icue_change_fan_end_icue" here)

## 4- Data structure also differs from driver assumptions
Once past the dtype check, the data returned by endpoints `0x60 0x6d` and `0x61 0x6d` is not the
simple `[device_count, val0, val1, ...]` array the driver parses. Both endpoints return a
fixed 9-byte opaque header (`09 00 00 00 01 ff [3 variable bytes]`) followed by the actual
values:

```
endpoint 0x60 0x6d (speed mode selector):
  res[0..8]  = opaque header
  res[9]     = speed mode byte — iCUE writes 0x01 for fixed percent, 0x02 for curve

endpoint 0x61 0x6d (fixed duty values):
  res[0..8]  = opaque header
  res[9]     = device_count (e.g. 0x06)
  res[10+n]  = duty for channel n — 1 byte (0–100), not 2-byte LE as driver assumes
```

The iCUE full write for endpoint '0x61 0x6d' with 6 fans at 1–6%:
```
dtype: 7e 20 | header: 09 00 00 00 01 ff ff ff ff | count: 06 | duty: 01 02 03 04 05 06
```

---
I'm working  on fixing the dtype constants, the data extraction offsets, and the duty encoding
before my next hardware test. Will update with results.

Do your **fw2.x captures** show the same `7e 20` dtype for these endpoints, or different?
That would help narrow down whether this is a fw-version difference or a model difference.

---

By all means, thanks again for the groundwork over the `_send_command`. The retry pattern was exactly the right fix for this stale responses issues.
_By the way - woah, analysing and parsing patterns over bytes is really some tiring activity..._

randomizedcoder, 2026-04-19:
Love it guys!

I stopped using the Commander that didn't work.  It's just sitting a draw, so I'd be happy to mail/post it somewhere if it would help fix this.  This did drive me nuts for a while, so this would saving some other person the pain I suffered.  ... The other one I got is working great.  I now have a fancy nix fan2go module that configures it and it works great.

OrcVole, 2026-08-15:
A datapoint rather than another report of the bug: on this machine the symptoms in this thread turned out to be **contention with another process holding the HID device**, not the firmware-2.x protocol problem. Posting it because CoolerControl is a very common way to drive these units, and separating the two causes may save some triage.

**Device and versions**

| | |
|---|---|
| Device | Corsair iCUE Commander CORE, `1b1c:0c32`, bundled with an H115i ELITE CAPELLIX |
| liquidctl | 1.16.0 |
| CoolerControl | 4.3.1 (`coolercontrold`) |
| OS / kernel | Fedora 44 (Bazzite), 7.1.5 |

**Symptom, which matches this thread**

```
$ liquidctl status
ERROR: Corsair Commander ST (broken): unexpected error:
       AssertionError('response does not match command')
```

Alongside it, the fans appeared stuck at a fixed speed regardless of load — the shape described in #750 and CoolerControl #381.

**With `coolercontrold` stopped, everything works — read and write**

```
$ sudo systemctl stop coolercontrold
$ liquidctl status
Corsair Commander ST (broken)
├── Pump speed           2297  rpm
├── Fan speed 1             0  rpm
├── Fan speed 2             0  rpm
├── Fan speed 3             0  rpm
├── Fan speed 4          1442  rpm
├── Fan speed 5          1461  rpm
├── Fan speed 6          1432  rpm
└── Water temperature    37.1  °C
```

Writes are honoured and are per-channel. `liquidctl set fan4 speed 100` took fan 4 from 1463 to 2046 rpm while fan 5 stayed at 1486. Ramping each port individually moved only its own fan:

| ramped | fan4 | fan5 | fan6 |
|---|---|---|---|
| port 4 | **1545** | 1072 | 1052 |
| port 5 | 1285 | **1530** | 1055 |
| port 6 | 1058 | 1244 | **1531** |

Driving all three at once gave 2221 / 2264 / 2222. No `ExpectationNotMet`, no `AssertionError`, in any of it.

**The "stuck fans" half had a second, unrelated cause worth mentioning.** Our CoolerControl config was applying settings to `fan1`, `fan2` and `fan3` — three ports that are physically empty on this unit — while `fan4`, `fan5` and `fan6`, which carry the actual cooler fans, had no setting at all and therefore sat at the firmware's floor speed. That looks exactly like "fans stuck and won't respond" from the user's side, and it is a configuration fault rather than a driver one. Pointing settings at the populated channels fixed it completely; the fans now follow a curve.

**Suggested disambiguation for other reporters in this thread**

```
sudo systemctl stop coolercontrold    # or quit any other process using the device
liquidctl status
liquidctl set fanN speed 100          # N = a port you know has a fan on it
```

If it works with the daemon stopped, the device is fine and the problem is elsewhere.

**What this does not explain.** It does not account for @rathko's finding that the failure reproduces under read-only `status` traffic and that a raw `/dev/hidraw` write+read bypassing hidapi also gets no response during the failure window. That is clearly a different cause with the same signature, and nothing here contradicts it — we simply did not reproduce a wedged device in these tests, including while issuing repeated `set` calls.

**One small suggestion, if it's welcome.** Since exclusive access by another process produces an error that reads like a protocol mismatch, it might be worth detecting a busy/held HID device and saying so — something like "device is in use by another process" — rather than surfacing `AssertionError('response does not match command')`. Judging by this thread, a fair number of reports may be this rather than firmware 2.x.

Happy to run further tests on this unit if any would help.

iamdadmin, 2026-09-13:
I'm assuming I'm subject to the device hang/reset symptom, but I'm getting a different error message than the above, so wanted to call it out in case it helps.

```
ERROR coolercontrold::engine::commanders::graph] Error applying Graph/Mix Profile calculated duty - Error on CommanderCore:fan4 for duty 50 - Setting fixed speed for LIQUIDCTL Device #2: a32b0548f1a0a44e50607647a9536e2fbb206093f6aeafaad871ae15e12ec87e - Liqctld Request failed with status:502 Bad Gateway - Body: Unexpected Device communication error: unspecified liquidctl error
```

Here's a little debug info. Didn't want to cloud the issue if it's not related, but feel free to @ me if I can provide useful info.

```
liquidctl --match core status --debug
[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.16.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-7.2.4-3-cachyos-x86_64-with-glibc2.44
[DEBUG] (cli) (_log_env_infos): python: 3.14.7 (main, Aug 14 2026, 06:38:32) [GCC 16.2.1 20260810]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.10.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.3.1
[DEBUG] (cli) (_log_env_infos): with pillow: 12.3.0
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
(...snip...)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core XT (broken)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0002)
(...snip...)
```

matteodevenuto, 2026-10-12:
Data point from a Commander CORE (`1b1c:0c1c`) on firmware **2.11.221**, liquidctl 1.16.0, Linux 7.2.

**Reads only work in software mode.** With no WAKE (`01 03 00 02`) first, `OPEN_ENDPOINT` (`0d 00 <mode>`) returns status `0x03` in byte 2. The following read then has the wrong data type and fails with `ExpectationNotMet`. After WAKE, speeds, temperatures, LED counts (`0x20`) and the port config (`0x1e`) all read correctly.

**Property reads work in either mode.** `02 <property>` succeeds in hardware mode. Property `0x03` reports the current mode (`01` = hardware, `02` = software), which gives a cheap way to check state. I scanned properties `0x00`–`0xff` in hardware mode. Apart from firmware (`0x13`) and VID/PID (`0x11`/`0x12`), none of them hold speed or temperature data.

**The controller falls back to hardware mode on its own.** After a WAKE with no SLEEP, a read 5 s later still found it in software mode. A read 35 s later found it back in hardware mode. Sending traffic every 3 s kept it in software mode for the full 90 s I tested.

**Direct LED writes work on 2.11.** This uses the same format as OpenRGB's controller:
- open endpoint `0x22`
- write with data type `12 00`
- RGB triplets: AIO zone first (29 LEDs here), then each RGB port padded to 34 LEDs
- close the endpoint

A full frame takes about 40 ms of USB time.

Together these explain the RGB flash from #448 and point to the only workaround I can see. Any one-shot command has to WAKE, and the controller returns to hardware mode on its own afterwards. The effect restarts either way, whether liquidctl sends SLEEP or the timeout does. Avoiding the flash needs something that stays running, keeps sending traffic and drives the LEDs itself. I wrote a small daemon that does this on top of liquidctl's driver: https://github.com/matteodevenuto/omarchy-corsair-cooling.

**Speed writes work on 2.11, but stored curves only run in hardware mode.** `set fans speed <curve>` and `set pump speed <duty>` work as long as WAKE comes first. Reading back `0x60`/`0x61`/`0x62` shows the stored values match what was written. The controller only follows them in hardware mode, though. In software mode the fans and pump sit at a software default (about 50% fans here) until the host sets their duty. Duty is set through endpoint `0x18`, data type `07 00`, with payload `[n_channels, (channel, 0x00, percent, 0x00) × n]`. I used the same sequence as FanControl.CorsairLink: close/open/write/close on handle `0xfc`. The pump responds right away, and the fans ramp toward the target at about 50 rpm/s. Because of the hardware-mode limitation, a liquidctl curve on its own only behaves as expected because SLEEP returns the controller to hardware mode.