liquidctl/liquidctl #754: Commander Core XT fan curve operation not supported by the driver

Eli2H, 2024-12-22:
### Describe the bug

Setting fixed speeds works, but when trying to set a curve the operation not supported error is thrown. This also happens when not specifying a temperature sensor.

### Commands executed

$ liquidctl --match Commander --temperature-sensor 1 set fans speed 20 20  30 35  40 55  52 90

### Output of all relevant commands with `--debug` flag

$ liquidctl --debug --match Commander --temperature-sensor 1 set fans speed 20 20  30 35  40 55  52 90
[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.13.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.12.4-arch1-1-x86_64-with-glibc2.40
[DEBUG] (cli) (_log_env_infos): python: 3.12.7 (main, Oct  1 2024, 11:15:50) [GCC 14.2.1 20240910]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.8.2
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1
[DEBUG] (cli) (_log_env_infos): with pillow: 11.0.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0052 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0050 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-17
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0037 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-15
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0053 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0051 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-18
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-16
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: 0-0036 (skipped, not a bus or unsupported adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-14
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c0b (usage_page=0xffc0 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0001)
[DEBUG] (usb) (probe): CommanderCore identified: Corsair Commander Core XT (experimental)
[DEBUG] (usb) (find_devices): HID device: 1b1c:0c2a (usage_page=0xff42 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 26ce:01a2 (usage_page=0x0001 usage=0x0000)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0013)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x000d usage=0x0005)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x000d usage=0x0022)
[DEBUG] (usb) (find_devices): HID device: 046d:c548 (usage_page=0x0001 usage=0x0022)
[... 59 lines cut ...]
    dev.set_speed_profile(args['<channel>'].lower(), profile, **opts)
  File "/usr/lib/python3.12/site-packages/liquidctl/driver/commander_core.py", line 140, in set_speed_profile
    raise NotSupportedByDriver
liquidctl.error.NotSupportedByDriver: operation not supported by the driver
[ERROR] (cli) (log): Corsair Commander Core XT (experimental): operation not supported by the driver

### Affected device

Corsair Commander Core XT

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Arch Linux

### Installation method

Official Arch Linux Repository

### Version of liquidctl

liquidctl v1.13.0 (Linux-6.12.4-arch1-1-x86_64-with-glibc2.40)

jonasmalacofilho, 2024-12-29:
Yeah, as indicated, that feature isn't currently supported.

closed 2024-12-29 by jonasmalacofilho

Eli2H, 2024-12-29:
> Yeah, as indicated, that feature isn't currently supported.

Oh okay, thought it was because it is on the core, core xt, core st page.

jonasmalacofilho, 2024-12-29:
That's a fair point. The device guides here on GitHub reflect the state of the code also here on GitHub (and on the same branch). And in this specific case, the feature hasn't yet made it to a release.

For the future, these cases are indicated in the documentation with "New in git".

jonasmalacofilho, 2024-12-29:
Also note that the feature in git may have some issues, as reported in #705. Contributions to improve this situation are very much welcome.