liquidctl/liquidctl #756: Add support for NZXT Kraken 360 [1e71:300e]

qbarnes, 2025-01-04:
### Device type

AIO liquid cooler

### Product page

https://nzxt.com/product/kraken-360

### First-party software

NZXT Cam

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles

### Physical connection

USB header

### Connection protocol

USB

### Additional information about the device

```
$ sudo lsusb -v -d 1e71:300e

Bus 001 Device 004: ID 1e71:300e NZXT NZXT Kraken Base
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x300e NZXT Kraken Base
  bcdDevice            1.00
  iManufacturer           1 NZXT Inc.
  iProduct                2 NZXT Kraken Base
  iSerial                 3 =ú^¼vpR¹!,5
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0039
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          4 In Application
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               5
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     619
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      0
Device Status:     0x0000
  (Bus Powered)
```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux

jonasmalacofilho, 2025-01-04:
Check liquidctl 1.14.0, this device should already be supported (modulo some issues with recent firmware, but I'm not sure those affect this particular model or not).

closed 2025-01-04 by jonasmalacofilho

qbarnes, 2025-01-04:
You're correct!  I just built 1.14.0 directly from this repo and it works for me.  Apologies for wasting your time.

qbarnes, 2025-01-04:
Anyone looking for the updated liquidctl-1.14.0 RPMs for Fedora, you can snag them from https://bodhi.fedoraproject.org/updates/FEDORA-2025-ab98447500.

jonasmalacofilho, 2025-01-04:
> You're correct! I just built 1.14.0 directly from this repo and it works for me. Apologies for wasting your time.

Glad to hear, and don't worry about the rest. Thanks for the feedback!

> If you have tests you'd like me to run to check firmware compatibility, just let me know.

On second glance, it seems that the only known issue is that setting the screen to display a GIF is not supported on firmware 2.x (#631).