liquidctl/liquidctl #758: MSI MEG CORELIQUID S360

aorozco-g, 2025-01-12:
### Device type

AIO LIquid cooler

### Product page

https://msi.com/Liquid-Cooling/MEG-CORELIQUID-S360

### First-party software

MSI Center

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, voltages, current or power limits, lighting of embedded LEDs

### Physical connection

USB header

### Connection protocol

USB

### Additional information about the device

> sudo lsusb -v
...

```
Bus 001 Device 013: ID 0db0:6a05 Micro Star International MEG Core Liquid S360
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0db0 Micro Star International
  idProduct          0x6a05 MEG Core Liquid S360
  bcdDevice            1.00
  iManufacturer           1 MSI
  iProduct                2 MEG Core Liquid S360
  iSerial                 3 A02021031800
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0040
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength     154
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux, attempting to capture USB/HID traffic

marrufa, 2025-04-16:
can be useful for you too ;)
https://github.com/liquidctl/liquidctl/pull/784

jonasmalacofilho, 2025-11-15:
We'll no longer taking new device requests. Instead, please see [how](https://github.com/liquidctl/liquidctl/blob/main/CONTRIBUTING.md) you can help add support for that device and contribute your changes back to the project.

closed 2025-11-15 by jonasmalacofilho