liquidctl/liquidctl #759: ITE 5711 Controller

xorbetto, 2025-01-18:
Found this controller on gigabyte aorus ultra z890i.
Simply add 0x048d, 0x5711 to rgb_fusion2.py driver make it works.

_MATCHES = [
        (0x048d, 0x5702, 'Gigabyte RGB Fusion 2.0 5702 Controller', {}),
        (0x048d, 0x8297, 'Gigabyte RGB Fusion 2.0 8297 Controller', {}),
        (0x048d, 0x5711, 'Gigabyte RGB Fusion 2.0 5711 Controller', {}),
    ]

clarin-ebtio800090, 2025-01-18:
Hello, we tried to solve the issue.

This is what we did:

Added the 0x048d, 0x5711 device to the _MATCHES list in the RgbFusion2 class to support the Gigabyte Aorus Ultra Z890i motherboard.

You can review changes in this commit: https://github.com/clarin-ebtio800090/liquidctl-liquidctl-759/commit/6de49e6c6f0182f12254807409a67fb23df0c6f9.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

jonasmalacofilho, 2025-01-18:
@xorbetto,

Thanks!

Can you confirm or check that all features documented in the [device guide](https://github.com/liquidctl/liquidctl/blob/main/docs/gigabyte-rgb-fusion2-guide.md) work?

jbbandos, 2025-03-04:
@jonasmalacofilho for me (aorus pro X870e) it works - but I need to always pass the parameter "-n 7" as it somehow detects 7 "ghost" devices, 0 to 6, and the real one as 7. No idea why, as a lsusb only shows a single entry for the ITE 5711.

jonasmalacofilho, 2025-03-04:
That suggests multiple HID usages are being found. The driver already has [some code](https://github.com/liquidctl/liquidctl/blob/e1d0b608d01f74b39c2be947db6f63ecfa5a7ba6/liquidctl/driver/rgb_fusion2.py#L107-L126) to automatically pick the correct one, but apparently that isn't working terribly well.

Can you paste the output of `liquidctl list -g`?

jbbandos, 2025-03-05:
Sure, I'll add it here. You'll notice that the working device is the one at /dev/hidraw6, and all others are at /dev/hidraw5. I've tested a bit more, and have to apologize, but the MB only turns off the rgb leds when I issue a liquidctl command, the AIO (a ICEMYST 360) is the only one that obeys the commands and shows colours, cycles, flashes or pulses.
```
~# liquidctl list -g
[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 0.0.0-unknown
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.13.5-1-liquorix-amd64-x86_64-with-glibc2.36
[DEBUG] (cli) (_log_env_infos): python: 3.11.2 (main, Nov 30 2024, 21:22:50) [GCC 12.2.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 default, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.9.0.post3
[DEBUG] (cli) (_log_env_infos): with pillow: 9.4.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1.post2
[DEBUG] (cli) (_log_env_infos): with smbus: version n/a (No package metadata was found for smbus)
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-ITE8800:00 (skipped, invalid literal for int() with base 10: 'ITE8800:00')
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-15
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-ITE8800:01 (skipped, invalid literal for int() with base 10: 'ITE8800:01')
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-16
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-14
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 046d:4096 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4096 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4096 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4096 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4096 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0001)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 8297 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0002)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 8297 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0020)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 8297 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0022)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 8297 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0050)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 8297 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0060)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 8297 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0070)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 8297 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 8297 Controller
[DEBUG] (usb) (find_devices): HID device: 3434:02a1 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 3434:02a1 (usage_page=0xff60 usage=0x0061)
[DEBUG] (usb) (find_devices): HID device: 3434:02a1 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 3434:02a1 (usage_page=0x0001 usage=0x0001)
[... cut ...]
├── Product ID: 0x5711
├── Release number: 0x0003
├── Bus: hid
├── Address: /dev/hidraw5
└── Driver: RgbFusion2
[DEBUG] (cli) (_list_devices_human): MRO: RgbFusion2, UsbHidDriver, BaseUsbDriver, BaseDriver, object

Device #7: Gigabyte RGB Fusion 2.0 8297 Controller
├── Vendor ID: 0x048d
├── Product ID: 0x5711
├── Release number: 0x0003
├── Bus: hid
├── Address: /dev/hidraw6
└── Driver: RgbFusion2
[DEBUG] (cli) (_list_devices_human): MRO: RgbFusion2, UsbHidDriver, BaseUsbDriver, BaseDriver, object

jonasmalacofilho, 2025-03-05:
Can you give #771 a try?

jonasmalacofilho, 2025-03-05:
I'm not sure what you meant by:

> [...] have to apologize, but the MB only turns off the rgb leds when I issue a liquidctl command, the AIO (a ICEMYST 360) is the only one that obeys the commands and shows colours, cycles, flashes or pulses.

What commands and channels are specifying in each case? And what was the expected behavior?

jbbandos, 2025-03-05:
For initialize:
```
~# liquidctl initialize
ERROR: too many devices, filter or select one (see: liquidctl --help)
~# liquidctl -n 7 initialize
Gigabyte RGB Fusion 2.0 8297 Controller
├── Hardware name       IT5711-GIGABYTE V1.0.18.5  
└── Firmware version                     1.0.18.5  

~# liquidctl -n 0 initialize
ERROR: Gigabyte RGB Fusion 2.0 8297 Controller: unexpected OS error: OSError('read error')
```
Then only for "-n 7" I did a few commands to test:
```
~# 
~# liquidctl -n 7 set sync color color-cycle --speed ludicrous
~# liquidctl -n 7 set led1 color fixed 350017
~# liquidctl -n 7 set led2 color fixed 350017
~# liquidctl -n 7 set led2 color fixed 350013
~# liquidctl -n 7 set led3 color fixed 350013
~# liquidctl -n 7 set led4 color fixed 350013
~# liquidctl -n 7 set led5 color fixed 350013
~# liquidctl -n 7 set led6 color fixed 350013
~# liquidctl -n 7 set led7 color fixed 350013
```
"led7" or "sync" applies to the AIO, all other rgb leds are off. "led" 1 through 6 and 8 don't do anything, all MB leds are off. I also tried pulse and flash commands, same result.

jbbandos, 2025-03-05:
> Can you give [#771](https://github.com/liquidctl/liquidctl/pull/771) a try, regarding the detection of (a single instance) the device?

The output of "liquidctl list -g" with the changes from pull #771 
```~# liquidctl list -g
[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 0.0.0-unknown
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.13.5-2-liquorix-amd64-x86_64-with-glibc2.36
[DEBUG] (cli) (_log_env_infos): python: 3.11.2 (main, Nov 30 2024, 21:22:50) [GCC 12.2.0]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 default, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.7.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.9.0.post3
[DEBUG] (cli) (_log_env_infos): with pillow: 9.4.0
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.2.1.post2
[DEBUG] (cli) (_log_env_infos): with smbus: version n/a (No package metadata was found for smbus)
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-ITE8800:00 (skipped, invalid literal for int() with base 10: 'ITE8800:00')
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-15
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-ITE8800:01 (skipped, invalid literal for int() with base 10: 'ITE8800:01')
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-16
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-14
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AuraLed, CommanderCore, CommanderPro, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 046d:4096 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4096 (usage_page=0x0001 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4096 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:4096 (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:4096 (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046d:c52b (usage_page=0xff00 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0001)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5711 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0002)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5711 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0020)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5711 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0022)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5711 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0050)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5711 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0060)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5711 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0x0059 usage=0x0070)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5711 Controller
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0xff89 usage=0x0010)
[DEBUG] (usb) (find_devices): HID device: 048d:5711 (usage_page=0xff89 usage=0x00cc)
[DEBUG] (usb) (probe): RgbFusion2 identified: Gigabyte RGB Fusion 2.0 5711 Controller
[DEBUG] (usb) (find_devices): HID device: 3434:02a1 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 3434:02a1 (usage_page=0xff60 usage=0x0061)
[DEBUG] (usb) (find_devices): HID device: 3434:02a1 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 3434:02a1 (usage_page=0x0001 usage=0x0001)
[... 80 lines cut ...]
├── Product ID: 0x5711
├── Release number: 0x0003
├── Bus: hid
├── Address: /dev/hidraw0
└── Driver: RgbFusion2
[... cut ...]
├── Product ID: 0x5711
├── Release number: 0x0003
├── Bus: hid
├── Address: /dev/hidraw0
└── Driver: RgbFusion2
[DEBUG] (cli) (_list_devices_human): MRO: RgbFusion2, UsbHidDriver, BaseUsbDriver, BaseDriver, object

Device #7: Gigabyte RGB Fusion 2.0 5711 Controller
├── Vendor ID: 0x048d
├── Product ID: 0x5711
├── Release number: 0x0003
├── Bus: hid
├── Address: /dev/hidraw1
└── Driver: RgbFusion2
[DEBUG] (cli) (_list_devices_human): MRO: RgbFusion2, UsbHidDriver, BaseUsbDriver, BaseDriver, object
```

jonasmalacofilho, 2025-03-05:
Regarding automatically filtering the right device, the problem is that we need to support old HIDAPI versions where the usage and usage page field are left uninitialized.

Because of this, we can't easily differentiate when a usage/usage page pair is garbage (which means we want to initialize the driver, actually) from when it's not garbage but also not what we want.

The driver currently has a workaround for it, and I tried to extend it a bit to cover this new device. I messed up, and forgot to consider the second usage page (0x0059). But I also don't like continuing to grow this hack...

jonasmalacofilho, 2025-03-06:
@xorbetto what was your experience with the driver on this device? You didn't mention any issues, so maybe you had a better time than @jbbandos (possibly due to different dependency versions).

TsunamiMommy, 2025-04-07:
I just wanted to drop this here as it is relevant information for properly supporting this device. https://github.com/developersu/argbColors/blob/master/SPECIFICATION.md This is a basic writeup of the controller specification along with a simple program to control it. I'll be doing testing on it once the rest of my new hardware comes in.

TsunamiMommy, 2025-04-13:
I've updated OpenRGB with support for the it5711. See https://gitlab.com/CalcProgrammer1/OpenRGB/-/merge_requests/2810 and https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/4306 . Let me know if you need additional information as I've added support for the ARGBV2.3 header along with some other basic information.

TsunamiMommy, 2025-04-13:
what do you mean? This is not the place for that kind of thing. This is merely trying to get it5711 support enabled in liquidctl. If you want to know other stuff then check the described link as there is where you should be asking that question. That or the openrgb discord.

awdsw, 2025-04-13:
> что вы имеете в виду? Это не место для таких вещей. Это просто попытка включить поддержку it5711 в liquidctl. Если вы хотите узнать что-то еще, то проверьте указанные ссылки, так как там вы должны задать этот вопрос. Это или openrgb discord. Чтобы заставить поддержку it5711 работать на Windows, вам нужно будет проверить последний релиз и pull request 2810 и собрать его для Windows. Если вы не знаете, как это сделать, то подождите, пока он будет объединен и выпущен.

Thanks, it's probably worth waiting! By the way, how long do you have to wait?

TsunamiMommy, 2025-07-25:
I've got this controller and the other ite controllers mostly mapped out (except for the it82950 found on h810m boards). I'm working on upstreaming it to openrgb but let me know if you want this support in liquidctl as I have technical information I created along with some basic decoding tools to set things up. Note that for proper layouts gigabyte uses in internal mapping system and the data is quite large.

mibofra, 2026-01-09:
Hello! Just to know, any update about this issue?

mibofra, 2026-01-09:
> [@mibofra](https://github.com/mibofra) Use OpenRGB with these controllers.

I have already seen that these controllers are supported via OpenRGB (and I have seen your marge request), but specifically wanted to know the state of this issue, for liquidctl, and what is missing to accomplish (and what I can test/help with, if possible).

It is nice, as support reached other apps, that can reach liquidctl too (and so, programs using liquidctl to manage specific devices, like coolercontrol, can have the rgb lights with other sensors and fans, in one single place).

Personally I use all off, so it does not bother me so much, but as I now have the same controller on the motherboard, if anything can be done to push support, if some help is needed.

As far I have understood, there are "just" two issues:

- automatically filtering the right HID device (without breaking support of other ones)
- control other leds that are not (a)rgb ports

Am I missing anything?

TsunamiMommy, 2026-01-09:
@mibofra There are a lot of things that have to be implemented to support these controllers properly. It's more complex than just adding another chip ID. If you want to add the support here, you can use the OpenRGB controller code as a reference. I've included some reference documentation here. Hopefully it helps.

[Gigabyte-RGB.zip](https://github.com/user-attachments/files/24519145/Gigabyte-RGB.zip)