liquidctl/liquidctl #765: NZXT Control Hub [1e71:2022]

KNobles, 2025-02-14:
### Device type

Digital RGB Lighting & Fan Speed Controller

### Product page

https://nzxt.com/en-BE/product/control-hub

### First-party software

NZXT CAM

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles, lighting of embedded LEDs, lighting of accessories like RGB fans or LED strips

### Physical connection

USB Header

### Connection protocol

USB

### Additional information about the device

sudo lsusb -v -d 1e71:2022

Bus 001 Device 005: ID 1e71:2022 NZXT NZXT Control Hub
Negotiated speed: High Speed (480Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1e71 NZXT
  idProduct          0x2022 NZXT Control Hub
  bcdDevice            1.00
  iManufacturer           1 NZXT, Inc.
  iProduct                2 NZXT Control Hub
  iSerial                 3 6F16EE008559F55AB270D9A91504F41A
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength     192
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               4
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      0
Device Status:     0x0001
  Self Powered

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

testing changes on Linux

jfranco09, 2025-04-17:
I'm facing the same issue, and as a work around I cloned the repo locally and applied this fix https://github.com/liquidctl/liquidctl/issues/304. After this, had to rebuild the project locally by running `python -m pip install . ` within the project root.

Here is the added device 

```
        (0x1e71, 0x2022, None, 'Digital RGB Lighting & Fan Speed Controller', {
            'speed_channel_count': 3,
            'color_channel_count': 6 
        }),
```
The only problem is that I'm not sure about the count for speed and color channels or where to check this. If someone could help with this, it would be helpful.

berkkarahan, 2025-08-14:
Do we have any more updates on this?

closed 2026-01-11 by jonasmalacofilho