liquidctl/liquidctl #766: Add setting multiple fans simultaneously to Aquacomputer Octo.

TsunamiMommy, 2025-02-23:
Hi there,

I was wondering if it would be possible to modify the behavior of liquidctl to allow setting multiple fans for the Aquacomputer Octo on the terminal. As it stands I can only set each fan individually and it will generate an error when I try to set multiple at the same time though I can manually iterate multiple fans with separate commands. Is there any reason that liquidctl can't internally when given multiple fans as arguments? 

Here is what I currently have to do to set multiple fan speeds via terminal:

`liquidctl -m aqua set fan1 speed 100 && liquidctl -m aqua set fan2 speed 100`

Here is what I'm requesting it to do.
`liquidctl -m aqua set fan1 speed 100 fan2 speed 100`

Here is the current error when you try the second method.
`ERROR: Aquacomputer Octo: operation not supported by the driver`

Let me know if there is a specific reason why this can't be done. I'm genuinely curious.

aleksamagicka, 2025-02-24:
I don't think there's a specific reason which would _completely disallow_ us to implement that (not like Python itself will stop us), but that would complicate argument parsing to a needless degree IMHO, for little benefit. If you want to set them both as a single command (which I understand), why not put them into a bash/bat script and just run that?

TsunamiMommy, 2025-02-24:
So does this limitation also apply to the api backend as well? Does it limit how the api can be used to set fan speeds for this particular device? I'm asking because I've been having issues with Cooler Control being able to set fan speeds on this device and am wondering if this limitation is to blame or not.

aleksamagicka, 2025-02-24:
> That said, could/should we add a fan pseudo-channel to that device? What do you think, @aleksamagicka?

We can do that, and IIRC in a single HID request. Does CC use that?

> So far it seems to be that CC is attempting to apply settings too quickly for the device to be able to keep up.

That's possible, and already [accounted for](https://github.com/liquidctl/liquidctl/blob/588112ceffd5520f8c5ee0ac9c0c2264c65795a5/liquidctl/driver/aquacomputer.py#L434-L435) in the hwmon code path; perhaps we should add it to `_set_fixed_speed_directly()` as well.

TsunamiMommy, 2025-02-24:
CC uses the `_set_fixed_speed_directly()` code path.
Here is a sample error being generated. 
The reason is because the hwmon path isn't currently supported by liquidctl due to missing `pwm_enable` on the `aquacomputer_d5next` kernel module with the octo. It instead defaults to direct access as a result.
```
ERROR    coolercontrol_liqctld.device_service - Error setting fixed speed:
: Traceback (most recent call last):
:   File "/usr/lib/python3/dist-packages/coolercontrol_liqctld/device_service.py", line 483, in set_fixed_speed
:     status_job.result(
:   File "/usr/lib/python3.12/concurrent/futures/_base.py", line 456, in result
:     return self.__get_result()
:            ^^^^^^^^^^^^^^^^^^^
:   File "/usr/lib/python3.12/concurrent/futures/_base.py", line 401, in __get_result
:     raise self._exception
:   File "/usr/lib/python3/dist-packages/coolercontrol_liqctld/device_executor.py", line 36, in run
:     result = self.fn(**self.kwargs)
:              ^^^^^^^^^^^^^^^^^^^^^^
:   File "/usr/local/lib/python3.12/dist-packages/liquidctl/driver/aquacomputer.py", line 504, in set_fixed_speed
:     self._set_fixed_speed_directly(channel, duty)
:   File "/usr/local/lib/python3.12/dist-packages/liquidctl/driver/aquacomputer.py", line 446, in _set_fixed_speed_directly
:     ctrl_settings = self.device.get_feature_report(_AQC_CTRL_REPORT_ID, report_length)
:                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
:   File "/usr/local/lib/python3.12/dist-packages/liquidctl/driver/usb.py", line 497, in get_feature_report
:     data = self.hiddev.get_feature_report(report_id, length)
:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
:   File "hidraw.pyx", line 394, in hidraw.device.get_feature_report
: OSError: read error
```

TsunamiMommy, 2025-02-24:
Okay I manually added the 200ms timer to `aquacomputer.py` and it fixed the issue. It just needs the additional time on the `_set_fixed_speed_directly()` routine just as it does on the `_set_fixed_speed_hwmon()` routine. There are likely other issues related to this and the original question I had. As it seems for a device like the octo supporting batch commands makes a lot of sense. Especially when setting them individually in rapid succession leads to problems like this. The only thing preventing this issue from occuring on the terminal is the inherent latency in running the command itself and displaying the output. So it's something to consider.

PR #767 (2025-02-24): aquacomputer: Sleep for 0.2s in direct code path
Fixes #766. Add a 0.2s delay in the aquacomputer driver when setting config directly (not through hwmon) to match the hwmon path behavior, to avoid confusing the device when multiple sets are done in quick succession.
closed 2025-02-24 by jonasmalacofilho