liquidctl/liquidctl #774: Failed to switch active bucket" error on NZXT Kraken Elite 2024 LCD update

kaan9700, 2025-03-09:
### Describe the bug

I am using a custom script that generates a 640x640 image for the LCD display of my NZXT Kraken Elite 2024 (v2) and uploads it via liquidctl. The image is an infographic showing temperatures. Therefore it runs in a loop that refreshes the image every n seconds.
From time to time 2 errors occur:

1. `failed to switch active bucket`:
The initial monitoring of the octopus is displayed for a short time (liquid temperature) 

2. `failed to setup bucket for data transfer`: 
The entire LCD display of the octopus goes black for a short time until it is replaced by the new image at the next period.

The errors are from the `kraken3.py` file, more precisely from the `_send_data` function.

### Commands executed

This is the code i am executing:
```
def set_kraken_screen(image_path):
    result = subprocess.run([
        "/home/user/liquidctl-venv/bin/liquidctl",
        "--match", "Kraken",
        "set", "lcd", "screen",
        "static", image_path
    ], capture_output=True, text=True)
    print(result)

def main():
    script_dir = os.path.dirname(os.path.abspath(__file__)) 
    image_path = os.path.join(script_dir, "infograph_custom.png")

    while True:
        cpu_usage = psutil.cpu_percent(interval=1.0)
        cpu_temp = get_cpu_temp()
        gpu_temp, gpu_usage = get_nvidia_stats()

        img = create_custom_layout(gpu_temp, cpu_temp, cpu_usage)
        img.save(image_path)

        try:
            set_kraken_screen(image_path)

        except Exception as e:
            print(">> Kraken failed to update")
            print(e)

        time.sleep(5)

```

### Output of all relevant commands with `--debug` flag

I started the script by executing this command:
```
sudo /home/user/liquidctl-venv/bin/python ~/monitoring/main.py
```

The logs looks like this:

```
[INFO] Starting data transfer on bucket index 3
...
[ERROR] Failed to switch active bucket
...
[ERROR] Failed to setup bucket for data transfer

### Affected device

NZXT Kraken Elite 2024

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Ubuntu 24.04.2 LTS

### Installation method

Installation via GitHub Repository

### Version of liquidctl

liquidctl v1.15.0.dev16+ge1d0b60 (Linux-6.11.0-19-generic-x86_64-with-glibc2.39)

relative23, 2026-09-16:
Update to my July comment: the "has not dropped into the bootloader since" part no longer holds. With the verified setup (no streaming into a refused bucket) our 2024 Elite RGB has wedged into its bootloader eight more times between 2026-07-28 and 2026-09-15, under a carousel of 90–180 uploads/h, each time idle 0–20 s after a successful upload. Not streaming into a refused bucket still seems right (#927), but on this device the sustained upload workload itself is enough, and I have no rate yet at which it does not happen eventually.

lukas-shawford, 2026-09-17:
I can add a field report here, if it helps.

I also have a NZXT Kraken 2024 Elite RGB (`1e71:3012`), running firmware version 1.2.12 (though liquidctl reports the firmware version as 1.2.0 due to an issue with reading the firmware version, as you already noted in #927).

Mine has been running stable for an entire month now without so much as a flicker, using a custom-generated image uploaded to the device every 2 seconds:

https://github.com/user-attachments/assets/fa9a95e5-ca1b-492e-8f0c-dae358ef2031

I'm running a patched version of liquidctl, using v1.16.0 as the baseline (as of commit 292dc561, which was latest at the time), and with the following modifications:

- "Safe buckets" patch, as informed by this issue and @relative23's [repo](https://github.com/relative23/kraken-lcd) (as of commit [7e1ddd2](https://github.com/relative23/kraken-lcd/commit/7e1ddd219511d9297ef57717ad98defc30b7c39e) in that repo, which is when I started messing around with my Kraken; the linked PR #927 did not exist at the time when I started my experiment). However, I have a slightly different implementation -- sharing the code below.
- Merged the following PRs into my local copy:
  - #915 (as of 381505f4)
  - #916 (as of dead8142)
  - #882 (as of d8f72b56).

Of the various modifications made to my patched version of liquidctl, the "safe buckets" patch was the primary one responsible for making the device run stable. With that said, the version of the patch I'm running is somewhat different than what's in the PR. It only ever uses 2 of the buckets, and switches back and forth between them (double-buffer technique).

Since you mentioned you're still observing the device enter into bootloader mode, I just wanted to share the version that has worked for me on identical hardware and firmware version, in case it helps.

Note: Claude Opus assisted with the implementation. "Janky" is the PC name on which I have the Kraken device installed, along with my custom version of liquidctl -- so any references to that are just to de-conflict any existing variable names in kraken3.py, and to make it clear it's my own custom code rather than stock liquidctl. Some of the changes below are not strictly related to the safe buckets patch itself, and more to do with logging adjustments for troubleshooting / collecting metrics while I was still in the experiment phase, so just ignore those (didn't get around to cleaning those up yet).

Here is the version of the "safe buckets" patch I have been running with ("monkey-patch" style, just appended to the end of `liquidctl/driver/kraken3.py`):

```python
# Fixes the LCD going black on roughly every other image upload on the
# NZXT Kraken 2024 Elite RGB (1e71:3012). Upstream bug:
#   https://github.com/liquidctl/liquidctl/issues/774
#
# Three defects in the stock KrakenZ3 bucket handling, all observed on janky:
#
#   1. A refused _setup_bucket is logged and then ignored. The driver streams
#      the whole image into a bucket the firmware never allocated, the final
#      _switch_bucket also fails, and set_screen() returns as if it worked.
#      The LCD goes black until some later upload happens to succeed. That is
#      the black-frame flicker, and it is why every failed upload in testing
#      still came back as HTTP 200 from CoolerControl.
#
#      Worse than cosmetic: streaming megabytes into an unallocated bucket is
#      reported to have twice wedged this exact model into its bootloader
#      (1e71:3011), where the LCD is dead and only cutting standby power for
#      ~30s recovers it. A reboot does not. So a refusal must stop the upload
#      *before* any bulk data goes out.
#
#   2. Once all 16 buckets are occupied, _find_next_unoccupied_bucket returns
#      -1 and the stock code falls back to bucket 0 -- which is usually the
#      bucket currently being displayed. Deleting and rewriting the live
#      bucket blanks the screen for the duration of the transfer.
#
#   3. The only cleanup path, _delete_all_buckets(), switches the LCD to the
#      firmware liquid view before deleting. That is the "reverts to the stock
#      liquid screen" behaviour.
#
# The fix is a plain double buffer. Two dedicated buckets at two fixed memory
# offsets; each upload writes the one that is *not* on screen, then switches to
# it. Nothing ever touches the live framebuffer, allocation never grows, and
# there is no periodic cleanup burst to go wrong -- every frame does exactly
# the same work.
#
# An earlier revision instead cycled all 16 buckets and reclaimed inactive ones
# when they ran out. That worked (200/200 uploads clean) but did a burst of 15
# delete commands every 15th frame, which was the last remaining source of
# periodic device-state churn. Fixed slots remove it, and remove the memory
# budgeting and slot-picking logic along with it.
#
# The wire protocol is unchanged. Only bucket choice, memory placement, and
# refusal handling differ from stock.

import inspect as _janky_inspect
import logging as _janky_logging
[... 203 lines cut ...]

        self._janky_occupied |= {target}
        self._janky_uploads += 1
        _JANKY_LOG.debug(
            "janky: frame %d -> slot %d at %dKiB (%dKiB)",
            self._janky_uploads, target, offset, dataSize,
        )
        if _JANKY_LOG_EVERY and self._janky_uploads % _JANKY_LOG_EVERY == 0:
            _JANKY_LOG.info(
                "janky: %d uploads, %d setup refusals, %d full clears, slot %d live",
[... cut ...]
    KrakenZ3._switch_bucket = _janky_switch_bucket
    KrakenZ3._delete_all_buckets = _janky_delete_all_buckets
    KrakenZ3._janky_claim_slots = _janky_claim_slots
    KrakenZ3._send_data = _janky_send_data

    # No banner here on purpose -- see _janky_send_data. The stand-down warning
    # above stays at import because WARNING still reaches stderr via logging's
    # last-resort handler when nothing else is configured, and a patch that
    # refused to load has no first upload to report from.

```