liquidctl/liquidctl #776: NZXT Kraken 360 RGB (1e71:300e, 1e71:2021)

mikeygrunst, 2025-03-14:
### Device type

AIO Liquid Cooler

### Product page

https://nzxt.com/product/kraken-360-rgb?srsltid=AfmBOormEx1Tpw9aB_ZinLYAvI5jinZvaVc-eFeMnZsh12facHBjSLNv

### First-party software

NZXT CAM (but not for linux)

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

lighting of accessories like RGB fans or LED strips, fan/pump speeds or duty cycles

### Physical connection

USB 2.0 header

### Connection protocol

HID

### Additional information about the device

**Product IDs:**  
- `1e71:300e` (NZXT Kraken Base - Cooling Control)  
- `1e71:2021` (NZXT RGB Controller - Lighting)  

**Issue:**  
- `liquidctl` does not detect or control this device.
- `lsusb` confirms both USB devices are recognized.
- `coolercontrol` software detects the Kraken 360 but cannot control RGB.

**Commands & Output:**
lsusb | grep 1e71
Bus 001 Device 006: ID 1e71:300e NZXT NZXT Kraken Base
Bus 001 Device 008: ID 1e71:2021 NZXT NZXT RGB Controller

sudo liquidctl list --debug
[DEBUG] (usb) (find_devices): HID device: 1e71:300e (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 1e71:2021 (usage_page=0xff00 usage=0x0001)
(can find the devices, but no driver. ASUS Aura LED controller recognized).

Thank you for considering adding further support for the Kraken 360 RGB!

aleksamagicka, 2025-03-14:
`300e` is supported: https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/kraken3.py#L587

What version of liquidctl do you have?

moonsquid9631, 2025-03-14:
Thank you for looking into this!

liquidctl --version
liquidctl v1.15.0.dev16+ge1d0b60 (Linux-6.11.0-19-generic-x86_64-with-glibc2.39)

moonsquid9631, 2025-03-15:
Thanks! I ran sudo liquidctl --version and discovered an installation issue (output: v0.0.0-unknown). Now liquidctl recognizes the device. I am using the latest git version. Would it be possible to add support for controlling the fan RGBs?

I can see the pump and fan controls: 
sudo ~/liquidctl-venv/bin/liquidctl status
ASUS Aura LED Controller
├── ARGB channels: 3      
└──  RGB channels: 1      

NZXT Kraken 2023
├── Liquid temperature    25.7  °C
├── Pump speed            1110  rpm
├── Pump duty               25  %
├── Fan speed              695  rpm
└── Fan duty                20  %

And I can see the LCD control
sudo ~/liquidctl-venv/bin/liquidctl initialize --match "NZXT Kraken 2023"
NZXT Kraken 2023
├── Firmware version    1.0.0  
├── LCD Brightness        100  %
└── LCD Orientation         0  °

But I dont think the color control is implemented.

From: 
# Available color channels and IDs for model Z coolers
_COLOR_CHANNELS_KRAKEN2023 = {}

Happy to help test.

DJCardona, 2025-03-27:
Hi, I'm new here (and total newb) and I have been trying to get my NZXT Kraken Elite 360 (Kraken Elite V2) LEDs to work as well. This is what I did: 

**Issue:** There are no channels assigned to the _COLOR_CHANNELS_KRAKEN2023 in the kraken3.py driver

**Fix:** In the kraken3.py driver, I copied the line from the _COLOR_CHANNELS_KRAKENX into the  _COLOR_CHANNELS_KRAKEN2023 line. There doesn't seem to be much (if any) disparity between the channels for the KRAKENX or KRAKENZ.

Baseline: _COLOR_CHANNELS_KRAKEN2023 = {}

Initial Change: _COLOR_CHANNELS_KRAKEN2023 = {"external": 0b001, "ring": 0b010, "logo": 0b100, "sync": 0b111}

**Test (_I assume you installed globally since you're using sudo to command liquidctl_):** 

sudo liquidctl --match "kraken" --direct-access set external color ff0000
sudo liquidctl --match "kraken" --direct-access set ring color ff0000
sudo liquidctl --match "kraken" --direct-access set logo color ff0000
sudo liquidctl --match "kraken" --direct-access set sync color ff0000

My assumption is that the channel for your device fans is 0b001; I could be wrong. 

**Results:** 

External = LED Pump Ring
Ring = Radiator Fan LEDs
Logo = Null
Sync = Pump Ring & Radiator Fan LED Sync

**Final Change:** 

_COLOR_CHANNELS_KRAKEN2023 = {"ring": 0b001, "fan": 0b010, "sync": 0b111}

So far, no issues with these changes. 

I hope this helps.

moonsquid9631, 2025-03-31:
Thank you for sending this! I modified as suggested (along with several versions of it). I am actually now installed in mamba environment. I am able to initialize and see the device. However, when I modify the kraken3.py file to 

"_COLOR_CHANNELS_KRAKEN2023 = {"ring": 0b001, "fan": 0b010, "sync": 0b111}"

Test: liquidctl --match "NZXT Kraken 2023" initialize

Output: ERROR: NZXT Kraken 2023: unexpected error: AssertionError('missing messages (attempts=12, missing=1)'). 

I also played around with the order of these arguments and swapped their code numbers. Removing these arguments again restored function/initialization of liquidctl. I saw that the Non-Elite versions of Krakens were previously giving this error and were described as "peculiar". I have Kraken 360 RGB (non-elite). (see https://github.com/liquidctl/liquidctl/issues/631).

Thanks again for your message and congrats you got yours working!

jonasmalacofilho, 2025-03-31:
@DJCardona, can you open a PR with a patch specific for the device you have?

(`_COLOR_CHANNELS_KRAKEN2023` is currently referenced by three different PIDs).

jonasmalacofilho, 2025-11-15:
We'll no longer taking new device requests. Instead, please see [how](https://github.com/liquidctl/liquidctl/blob/main/CONTRIBUTING.md) you can help add support for that device and contribute your changes back to the project.