liquidctl/liquidctl #785: farbwerk 366 not listed with liquidctl status

BeatScherrer, 2025-04-23:
### Describe the bug

liquidctl list shows
```
→  sudo liquidctl list
Device #0: ASUS Aura LED Controller
Device #1: Aquacomputer Octo
Device #2: Aquacomputer Farbwerk 360
```

but liquidctl status does not output farbwerk 360:
```
→  sudo liquidctl status
ASUS Aura LED Controller
├── ARGB channels: 3
└──  RGB channels: 1

Aquacomputer Octo
├── Fan 1 speed       1001  rpm
├── Fan 1 power       0.25  W
├── Fan 1 voltage    12.24  V
├── Fan 1 current     0.02  A
├── Fan 2 speed        976  rpm
├── Fan 2 power       0.13  W
├── Fan 2 voltage    12.24  V
├── Fan 2 current     0.01  A
├── Fan 3 speed       1007  rpm
├── Fan 3 power       0.12  W
├── Fan 3 voltage    12.24  V
├── Fan 3 current     0.01  A
├── Fan 4 speed        988  rpm
├── Fan 4 power       0.14  W
├── Fan 4 voltage    12.24  V
├── Fan 4 current     0.01  A
├── Fan 5 speed          0  rpm
├── Fan 5 power       0.00  W
├── Fan 5 voltage    12.24  V
├── Fan 5 current     0.00  A
├── Fan 6 speed        983  rpm
├── Fan 6 power       0.15  W
├── Fan 6 voltage    12.24  V
├── Fan 6 current     0.01  A
├── Fan 7 speed          0  rpm
├── Fan 7 power       0.00  W
├── Fan 7 voltage    12.24  V
├── Fan 7 current     0.00  A
├── Fan 8 speed          0  rpm
├── Fan 8 power       0.00  W
├── Fan 8 voltage    12.24  V
└── Fan 8 current     0.00  A
```

### Commands executed

```
liquidctl status
```

### Output of all relevant commands with `--debug` flag

```
liquidctl status --debug
[DEBUG] (cli) (_log_env_infos): script: /nix/store/4qqg9yzkig52j8gnp6zalwg6y15v9353-python3.12-liquidctl-1.15.0/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.15.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.14.2-x86_64-with-glibc2.40
[DEBUG] (cli) (_log_env_infos): python: 3.12.9 (main, Feb  4 2025, 14:38:38) [GCC 14.2.1 20250322]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.9.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[... 164 lines cut ...]
└── Fan 8 current     0.00  A

[DEBUG] (cli) (main): device: Aquacomputer Farbwerk 360
[INFO] (aquacomputer) (get_status): bound to aquacomputer_d5next kernel driver, reading status from hwmon
```

### Affected device

Aquacomputer Farbwerk 360

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

NixOS unstable

### Installation method

nixpkgs

### Version of liquidctl

liquidctl v1.15.0 (Linux-6.14.2-x86_64-with-glibc2.40)

jonasmalacofilho, 2025-05-16:
Hi,

Can you post the output from the following commands?

```
# liquidctl list -g
# liquidctl -m farbwerk status -g
# liquidctl -m farbwerk initialize -g
# liquidctl -m farbwerk status --direct-access -g
```

You can remove the redundant probing messages from the later commands, but please don't remove anything from the `[DEBUG] (cli) (main): device:` line forwards.

---

Setting LEDs is really not supported yet. Only sensor monitoring is available on Farbwerk for now. This should be made clearer in the documentation.

jonasmalacofilho, 2026-01-11:
Closing due to lack of feedback.

closed 2026-01-11 by jonasmalacofilho