liquidctl/liquidctl #787: Thermalright Grand Vision

CapnRyna, 2025-05-10:
### Device type

AIO liquid cooler

### Product page

https://thermalright.com/product/grand-vision-360-black-argb/

### First-party software

Thermalright TRCC water-cooling LED screen control software

### What monitoring functionality does the device support?

temperatures, fan/pump speeds or duty cycles

### What configuration functionality does the device support?

fan/pump speeds or duty cycles

### Physical connection

USB Header

### Connection protocol

USB

### Additional information about the device

``` 
> lsusb -v

Bus 003 Device 006: ID 87ad:70db ChiZhu Tech USBDISPLAY
Couldn't open device, some information will be missing
Negotiated speed: High Speed (480Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x87ad ChiZhu Tech
  idProduct          0x70db USBDISPLAY
  bcdDevice            4.07
  iManufacturer           1 ChiZhu Tech
  iProduct                2 USBDISPLAY
  iSerial                 3 559d86173a326a53
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      0
      iInterface              4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

### Can you help with implementing or testing the changes needed for liquidctl to support this device?

attempting to capture USB/HID traffic, testing changes on Linux

feelinfroggy88, 2025-05-21:
https://thermalright.com/product/frozen-warframe-pro-360-white-argb/

feelinfroggy88, 2025-05-21:
I had added the previous post because I believe I had mad a mistake on the product page info. 
Thanks.

closed 2025-11-15 by jonasmalacofilho