liquidctl/liquidctl #798: NZXT Fan+RGB Controller (3+6 channel) 1e71:2020 Can't Control LED

tbelway, 2025-06-16:
### Describe the bug

I am unable to control the RGB LEDs for the NZXT FAN+RGB Controller using liquidctl nor coolercontrol (which uses liquidctl). However, I am able to control fan speed.

### Commands executed

liquidctl set sync color fixed "rgb(255,0,0)" --debug

### Output of all relevant commands with `--debug` flag

```bash
/usr/lib/python3.13/site-packages/docopt.py:165: SyntaxWarning: invalid escape sequence '\S'
  name = re.findall('(<\S*?>)', source)[0]
/usr/lib/python3.13/site-packages/docopt.py:166: SyntaxWarning: invalid escape sequence '\['
  value = re.findall('\[default: (.*)\]', source, flags=re.I)
/usr/lib/python3.13/site-packages/docopt.py:207: SyntaxWarning: invalid escape sequence '\['
  matched = re.findall('\[default: (.*)\]', description, flags=re.I)
/usr/lib/python3.13/site-packages/docopt.py:456: SyntaxWarning: invalid escape sequence '\S'
  split = re.split('\n *(<\S+?>|-\S+?)', doc)[1:]
[DEBUG] (cli) (_log_env_infos): script: /usr/bin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.15.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.14.6-106.bazzite.fc42.x86_64-x86_64-with-glibc2.41
[DEBUG] (cli) (_log_env_infos): python: 3.13.3 (main, Apr 22 2025, 00:00:00) [GCC 15.0.1 20250418 (Red Hat 15.0.1-0)]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.9.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0.post3
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.3.1
[DEBUG] (cli) (_log_env_infos): with pillow: 11.1.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3 (i915 gmbus dpc)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10 (AMDGPU DM i2c hw bus 3)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1 (i915 gmbus vga)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8 (AMDGPU DM i2c hw bus 1)
[DEBUG] (smbus) (find_devices): I²C adapter: 14-0053 skipped: not a bus or unsupported adapter
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6 (AUX D/DDI D/PHY D)
[DEBUG] (smbus) (find_devices): I²C adapter: 14-0051 skipped: not a bus or unsupported adapter
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-13 (AMDGPU DM aux hw bus 1)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4 (i915 gmbus dpb)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11 (AMDGPU DM i2c hw bus 4)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2 (i915 gmbus panel)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0 (i915 gmbus ssc)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9 (AMDGPU DM i2c hw bus 2)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7 (AMDGPU DM i2c hw bus 0)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-14 (SMBus I801 adapter at f040)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5 (i915 gmbus dpd)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12 (AMDGPU DM aux hw bus 0)
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AsusRyujin, AuraLed, CommanderCore, CommanderPro, Coolit, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, MpgCooler, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 214e:0005 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 214e:0005 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 214e:0005 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 214e:0005 (usage_page=0x0001 usage=0x0004)
[DEBUG] (usb) (find_devices): HID device: 214e:0005 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 04d8:eed2 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 04d8:eed2 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 04d8:eed2 (usage_page=0x0001 usage=0x0080)
[DEBUG] (usb) (find_devices): HID device: 04d8:eed2 (usage_page=0x000c usage=0x0001)
[DEBUG] (usb) (find_devices): HID device: 04d8:eed2 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1e71:2020 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (__init__): has kernel driver: nzxt-smart2 (/sys/class/hidraw/hidraw0/device/hwmon/hwmon2)
[DEBUG] (usb) (probe): SmartDevice2 identified: NZXT RGB & Fan Controller (3+6 channels)
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 8087:8000
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 8087:8008
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (usb) (find_devices): USB device: 045b:0210
[DEBUG] (usb) (find_devices): USB device: 045b:0210
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 045b:0209
[DEBUG] (usb) (find_devices): USB device: 8087:0032
[DEBUG] (usb) (find_devices): USB device: 1e71:2020
[DEBUG] (usb) (find_devices): USB device: 04d8:eed2
[DEBUG] (usb) (find_devices): USB device: 04d8:eec5
[DEBUG] (usb) (find_devices): USB device: 214e:0005
[DEBUG] (usb) (find_devices): USB device: 05e3:0610
[DEBUG] (usb) (find_devices): USB device: 045b:0209
[DEBUG] (usb) (find_devices): USB device: 1d6b:0002
[DEBUG] (cli) (main): device: NZXT RGB & Fan Controller (3+6 channels)
[DEBUG] (usb) (write): writing report 0x28 with 63 bytes: 03:3f:00:00:02:00:00:01:00:00:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
```

### Affected device

NZXT RGB and FAN controller

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Bazzite

### Installation method

Bazzite ujust

### Version of liquidctl

v1.150.0

tbelway, 2025-06-25:
Any other info I can provide to assist with this let me know.

jonasmalacofilho, 2025-06-25:
That feature isn't currently supported by our driver.

> In 2022 NZXT released a new version of the RGB & Fan Controller with 3 fan speed channels and 6 lighting channels.
>
> For now, this new version is only partially supported by liquidctl: only monitoring and fan control are available.
>https://github.com/liquidctl/liquidctl/blob/main/docs/nzxt-hue2-guide.md#nzxt-rgb--fan-controller-36-channels

tbelway, 2025-06-25:
> That feature isn't currently supported by our driver.

Oh shoot, I thought that was only for the 1e71:2022.
Sorry about that! Any way I can assist with adding this to the driver for a subsequent version?