liquidctl/liquidctl #800: Lights/colors not exposed on Kraken 2024 Elite RGB

PandorasFox, 2025-06-20:
### Describe the bug

When I initialize my kraken, it appears to not initialize any of the color channels.

I'm not really quite sure why this is the case - I did see one mention of firmware 1.2.1 somewhere, and I have firmware 1.2.0, so if there's a minimum fw version I can update the documentation/support matrix after figuring out the issue here - I'm just at a bit of a loss.

### Commands executed

```
[root@cerberus tartarus]# liquidctl --debug initialize

[root@cerberus tartarus]# liquidctl --debug set ring color fixed '0xffffff'

```

### Output of all relevant commands with `--debug` flag

<details>
<summary>Logs</summary>

```
[root@cerberus tartarus]# liquidctl --version
liquidctl v1.15.0 (Linux-6.14.9-300.fc42.x86_64-x86_64-with-glibc2.41)

[root@cerberus tartarus]# liquidctl --debug initialize
[DEBUG] (cli) (_log_env_infos): script: /usr/sbin/liquidctl
[DEBUG] (cli) (_log_env_infos): version: 1.15.0
[DEBUG] (cli) (_log_env_infos): platform: Linux-6.14.9-300.fc42.x86_64-x86_64-with-glibc2.41
[DEBUG] (cli) (_log_env_infos): python: 3.13.3 (main, Apr 22 2025, 00:00:00) [GCC 15.0.1 20250418 (Red Hat 15.0.1-0)]
[DEBUG] (cli) (_log_env_infos): encoding: UTF-8 current, UTF-8 preferred, utf8_mode 0
[DEBUG] (cli) (_log_env_infos): with colorlog: 6.9.0
[DEBUG] (cli) (_log_env_infos): with crcmod: 1.7
[DEBUG] (cli) (_log_env_infos): with docopt: 0.6.2
[DEBUG] (cli) (_log_env_infos): with hidapi: 0.14.0.post3
[DEBUG] (cli) (_log_env_infos): with pyusb: 1.3.1
[DEBUG] (cli) (_log_env_infos): with pillow: 11.1.0
[DEBUG] (cli) (_log_env_infos): with smbus: 1.1
[DEBUG] (cli) (_log_env_infos): with winusbcdc: version n/a (No package metadata was found for winusbcdc)
[DEBUG] (cli) (_log_env_infos): with libusb-package: version n/a (No package metadata was found for libusb-package)
[DEBUG] (smbus) (find_devices): searching LinuxI2c
[DEBUG] (smbus) (find_devices): LinuxI2c drivers: Ddr4Temperature, EvgaPascal, RogTuring, VengeanceRgb
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-3 (i915 gmbus dpb)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-10 (i915 gmbus tc6)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-1 (Synopsys DesignWare I2C adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: 12-0052 skipped: not a bus or unsupported adapter
[DEBUG] (smbus) (find_devices): I²C adapter: 12-0050 skipped: not a bus or unsupported adapter
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-8 (i915 gmbus tc4)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-6 (i915 gmbus tc2)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-4 (i915 gmbus dpc)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-11 (AUX USBC2/DDI TC2/PHY C)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-2 (i915 gmbus dpa)
[DEBUG] (smbus) (find_devices): I²C adapter: 12-0053 skipped: not a bus or unsupported adapter
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-0 (Synopsys DesignWare I2C adapter)
[DEBUG] (smbus) (find_devices): I²C adapter: 12-0051 skipped: not a bus or unsupported adapter
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-9 (i915 gmbus tc5)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-7 (i915 gmbus tc3)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-5 (i915 gmbus tc1)
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-MSFT8000:00 skipped: invalid literal for int() with base 10: 'MSFT8000:00'
[DEBUG] (smbus) (find_devices): I²C adapter: i2c-12 (SMBus I801 adapter at efa0)
[DEBUG] (usb) (find_devices): searching HidapiBus
[DEBUG] (usb) (find_devices): HidapiBus drivers: Aquacomputer, AsusRyujin, AuraLed, CommanderCore, CommanderPro, Coolit, CorsairHidPsu, H1V2, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, MpgCooler, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _BaseSmartDevice
[DEBUG] (usb) (find_devices): HID device: 046b:ff10 (usage_page=0x0001 usage=0x0002)
[DEBUG] (usb) (find_devices): HID device: 046b:ff10 (usage_page=0x0001 usage=0x0006)
[DEBUG] (usb) (find_devices): HID device: 1e71:3012 (usage_page=0xff00 usage=0x0001)
[DEBUG] (usb) (open): selected interface: 0
[DEBUG] (usb) (probe): KrakenZ3 identified: NZXT Kraken 2024 Elite RGB
[DEBUG] (usb) (find_devices): searching PyUsbBus
[DEBUG] (usb) (find_devices): PyUsbBus drivers: Hydro690Lc, HydroPro, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc
[DEBUG] (usb) (find_devices): USB device: 1d6b:0003
[DEBUG] (usb) (find_devices): USB device: 1e71:3012
[... 86 lines cut ...]
liquidctl.error.NotSupportedByDevice: operation not supported by the device
[ERROR] (cli) (log): NZXT Kraken 2024 Elite RGB: operation not supported by the device
```

</details>

### Affected device

NZXT Kraken 2024 Elite RGB

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

fedora server

### Installation method

dnf install liquidctl

### Version of liquidctl

liquidctl v1.15.0 (Linux-6.14.9-300.fc42.x86_64-x86_64-with-glibc2.41)

PandorasFox, 2025-06-20:
some time next week when I return, I'll attempt futzing with a local checkout + scripts using the apis for reproduction => debugging down the interfaces to see why the interfaces aren't initializing

btcthoty, 2025-08-05:
"ERROR: NZXT Kraken 2024 Elite RGB: operation not supported by the device" running in to same issue. please help

btcthoty, 2025-08-06:
> `git diff --no-index --ignore-all-space -- liquidctl/driver/kraken3.py /tmp/kraken3.py`
> 
> [@btcthoty](https://github.com/btcthoty), those changes are mostly pointless. They could appear to work if they end up bypassing the check for the feature (but I haven't checked that). In any case, they don't actually check that the device reported the new channels as available.
> 
> My unsolicited advice is: try to reason _for yourself_ about what the existing code does, where and why that doesn't work for this device, what are the necessary changes, and any possible side effects from them.
> 
> Also, a request: don't throw 0%-reasoning-100%-LLM-generated garbage ~around~ here. It's not that it can't eventually and just by chance "help" (by some definition of help). But you're making me[1](#user-content-fn-1-ddacd82f9c449541e3e76c525c34fc45) and possibly others read garbage, while we could be reading and discussing interesting things. Plus the internet is already filled with noise as it is.
> ## Footnotes
> 
>     1. As a maintainer I have to check your script to make sure it isn't something harmful. [↩](#user-content-fnref-1-ddacd82f9c449541e3e76c525c34fc45)

Thanks for your feedback on my submission (kraken3.zip). I shared this solution with the goal of helping others facing the same issue I documented—where liquidctl v1.15.0 fails to expose light/color controls for the NZXT Kraken 2024 Elite RGB, as evidenced by the NotSupportedByDevice error in my logs. I understand it might not align perfectly with the project’s current standards, but I see it as a starting point, not a final product. Open-source thrives on community contributions, and my intent was to offer a temporary workaround until full support can be implemented.
The error logs and my experience suggest a gap in either device support or documentation—e.g., the lack of clarity around firmware requirements (I’m on 1.2.0, and I’ve seen 1.2.1 mentioned elsewhere). This confusion affects users like me, and my solution could at least ease that frustration for others. If there are specific technical flaws—like compatibility risks or code quality issues—I’d welcome detailed feedback so I can refine it to meet your expectations. Collaboration is key here, and I’m happy to iterate.
That said, I’d also suggest that the tone of your reply could unintentionally discourage community involvement. A more constructive approach, even when rejecting a contribution, could encourage more people to engage with liquidctl. For example, pointing me to guidelines or suggesting next steps would turn this into a learning opportunity rather than a shutdown. I’d love to work together to address this bug and improve support for the Kraken 2024 Elite RGB—whether through my solution or another path. What do you think?

PandorasFox, 2025-08-06:
yeah, i have been struggling to find a concrete reason for why my device is not initializing (mostly from a lack of time directed towards it/never having had it work in the first place, so no point of comparison).

jonasmalacofilho, 2025-08-06:
@PandorasFox , from the logs your posted you're simply hitting a current (and hopefully temporary, see #803) limitation of the driver. The device appears to be initialized, but the driver doesn't yet know how to handle the ring LEDs.

btcthoty, 2025-08-07:
@jonasmalacofilho, your reply was a bit arrogant don't you think? Your Kraken guide mentions “Z-series support” for the Elite 360—ring, fans, static LCD images—and yet the driver literally skips most of it and only initialises fans. Please explain how "_COLOR_CHANNELS_KRAKEN2023 = {}" provides support for 2024 Elite?

All I did was open kraken3.py, yanked out the dead stub, pasted in five lines of elite’s USB IDs, and it worked—the ring glows and LEDs report perfectly. yet you still brushed it off as “AI garbage” without even bothering to run it. Funny how the moment someone drops in a working patch, it gets flagged as “AI-generated” and tossed aside without a single hardware test regardless if it was with the help of LLM's. 

Liquidctl is the only community alternative to NZXT’s bloated CAM, yet you keep shipping drip-feed releases and overhyping features that aren’t implemented. How about you own the gaps in the guide, merge fixes that prove they work (AI-bootstrapped or not), and push real updates more than once every six months? Until then, liquidctl will just be an empty promise with a community left banging on its door.

PandorasFox, 2025-08-07:
dawg it is straight up disrespectful of everyone else's time to post LLM output without further thought. it is a tool that demands further thought.

btcthoty, 2025-08-07:
@PandorasFox Disrespectful? How? All I did was point out broken functionality and offer a patch that works. Getting offended by an LLM output that provides a fix just shows how entitled you are.

PandorasFox, 2025-08-07:
> [@PandorasFox](https://github.com/PandorasFox) Disrespectful? How? All I did was point out broken functionality and offer a patch that works. Getting offended by an LLM output that provides a fix just shows how entitled you are.

i'm in the middle of writing my own systems software and am a seasoned engineer. far from being entitled to software, I know how annoying it is to field LLM output that has had no effort being interpreted by a human brain, nor cleaned up and conformed to the standards of the codebase.

the general culture of open source is that maintainers are generally donating _lots_ of time to projects they care about, and posting unreviewed LLM output at projects is, historically and personally speaking, generally a waste of time.

also it's.... not entitled to not want an un-human-reviewed hack fix touching the driver for the cooling pump on my home server.  The time spend looking at this and discussing it has been a net negative for everyone else involved, I think. Mentally, I don't really mind if issues take a while to get closed out; I prefer it if the notifications for an issue track actual progress towards an engineered fix so that I can get unblocked on my projects (such as writing a status-png generator that can show the status of what's going on on my home server). I feel bad adding more to this conversation at this point, tbh.

hyenatown, 2025-08-07:
Ironically enough the pytest suite for this driver doesn't work currently due to `pillow` (and, forgiving I didn't overlook how to fix this) means that automated testing for kraken3 is off the table, so, I'm trying to stay methodical and not make drastic (potentially breaking) changes.