liquidctl/liquidctl #818: Gigabyte RGB Fusion 2.0 5702: unable to turn off rgb lights

LAKostis, 2025-09-06:
### Describe the bug

Greetings!

Due some reason (maybe bug in liquidctl) it's not possible to turn off lights on Gigabyte RGB Fusion 2.0 5702. The board marked as supported and everything was detected but still light control doesn't work.

Environment

OS: ALT p11 x86_64
luquidctl version: liquidctl 1.15.0-alt0.1
Board: Gigabyte B760 DS3H RTL 

liquidctl list -g attached.

[liquidctl.log](https://github.com/user-attachments/files/22185466/liquidctl.log)

### Commands executed

```
# liquidctl initialize
Gigabyte RGB Fusion 2.0 5702 Controller
├── Hardware name       IT5701-GIGABYTE V3.5.5.0  
└── Firmware version                     3.5.5.0  
```

Trying to turn off the lights
`#  liquidctl set sync color off`
(and nothing happen)

### Output of all relevant commands with `--debug` flag

in attachment

### Affected device

Gigabyte RGB Fusion 2.0 5702

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

ALT Linux p11

### Installation method

Official ALTLinux repository

### Version of liquidctl

liquidctl 1.15.0

LAKostis, 2025-09-06:
I think it's board (B760 DS3H RTL) specific issue, as everything works just fine on B550 AORUS ELITE V2.

TsunamiMommy, 2025-09-10:
Don't use this driver for newer motherboards. Use OpenRGB instead. I've updated that driver to fully support these boards and this driver is very out of date. If somebody wants to fix this driver then they can use OpenRGB as a reference or @ me for protocol details.

LAKostis, 2025-11-10:
@TsunamiMommy , thanks for the tip, then I'm closing this issue as given workaround (use openrgb) works in my case.