liquidctl/liquidctl #826: Corsair HX1500i PSU reports: WARNING: some attributes cannot be read from corsair-psu kernel driver

rocketraman, 2025-09-29:
### Describe the bug

Corsair HX1500i PSU reports: WARNING: some attributes cannot be read from corsair-psu kernel driver

The message shows up when liquidctl is run.

lsusb output:

```

Bus 003 Device 005: ID 1b1c:1c1f Corsair HX1500i Power Supply
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1b1c Corsair
  idProduct          0x1c1f HX1500i Power Supply
  bcdDevice            1.00
  iManufacturer           1 CORSAIR
  iProduct                2 HX1500i Power Supply
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength      29
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
[... 159 lines cut ...]
├── +3.3V output power         3.00  W
├── Total power output       178.00  W
├── Estimated input power    200.00  W
└── Estimated efficiency         89  %
```

### Affected device

Corsair HX1500i

### Does your version of liquidctl support the device in question?

Yes, my version supports it

### Operating system and version

Fedora Linux 42

### Installation method

Official 

### Version of liquidctl

liquidctl v1.15.0 (Linux-6.16.7-200.fc42.x86_64-x86_64-with-glibc2.41)

rocketraman, 2025-09-30:
> Can you elaborate on what is wrong?

Thanks for your reply.

For context -- I have thousands of these in my system logs because it is output every time coolercontrold queries liquidctl for metrics.

> Is it how the warning is worded, or is more information available now from the kernel driver?

Two things:

1) When liquidctl is called with a structured format output like `--json` (which presumably tools like coolercontrold use), do not output the warning. Or alternatively, allow use of a new `--quiet` or `--nowarnings` option which suppresses it. That would allow tools like coolercontrold to use liquidctl without spamming the logs.

Even better, include the missing metrics in the JSON structure, maybe under a separate `missing` key for backward compat, so that tools have access to those as well.

2) The wording doesn't tell me anything useful. If there are specific metrics I should be seeing that I'm not because of a limitation in the kernel driver, then tell me what they are so I know what I'm missing.

A suggested approach for the default plain output would be to show the attribue with a `missing` value instead of emitting a warning to stderr. For example, instead of:

```
WARNING: some attributes cannot be read from corsair-psu kernel driver
Corsair HX1500i
…
```

the output could be something like:

```
Corsair HX1500i
├── VRM temperature            51.8  °C
…
├── Frobnitz              data missing
…```

jonasmalacofilho, 2025-09-30:
> 1. When liquidctl is called with a structured format output like `--json` (which presumably tools like coolercontrold use), do not output the warning. Or alternatively, allow use of a new `--quiet` [...]

AFAIK coolercontrol is using the liquidctl API, not the CLI (with or without `--json`). We're also emitting these warnings through `logging`, so coolercontrol can already filter them out (if they want).

> 2. The wording doesn't tell me anything useful. If there are specific metrics I should be seeing that I'm not because of a limitation in the kernel driver, then tell me what they are so I know what I'm missing.

I'll need to check, but it's possible that the vagueness has to do with differences in kernel driver versions...

> Even better, include the missing metrics in the JSON structure, maybe under a separate `missing` key for backward compat, so that tools have access to those as well.

That might be annoying for consumers, as they would have to test both if the field is present as well as if isn't null/"missing".

> A suggested approach for the default plain output would be to show the attribue with a `missing` value instead of emitting a warning to stderr. For example, instead of:

For normal, user-intended output, that's a reasonable suggestion.

However, if that means suppressing the warning, I'm concerned that users will then ask "why?", making us bring back the warning anyway.

rocketraman, 2025-09-30:
> AFAIK coolercontrol is using the liquidctl API, not the CLI (with or without `--json`). We're also emitting these warnings through `logging`, so coolercontrol can already filter them out (if they want).

Thanks. I've created issue https://gitlab.com/coolercontrol/coolercontrol/-/issues/508 to track this.

> That might be annoying for consumers, as they would have to test both if the field is present as well as if isn't null/"missing".

Agreed. Better to simply set the JSON metric value to `null` if the value is not available. Since `null` is valid in JSON, this should be fine, unless previous JSON-schema or other documentation has indicated the value will never be `null`.

> For normal, user-intended output, that's a reasonable suggestion.
>
> However, if that means suppressing the warning, I'm concerned that users will then ask "why?", making us bring back the warning [...]

Makes sense. It seems reasonable to output both the `missing` value and the warning by default, and allow users to suppress the warning explicitly via a CLI option.